rickysarraf/laptop-mode-tools #125: LMT or TLP, Powertop?

ManuGithubSteam, 2018-02-09:
Hi

Thanks for your hard work on this!

is there any difference between TLP and Laptop Mode Tools ?

Any situation in there TLP may save more battery juice ?

Should i use Laptop Mode tools and "powertop --auto-tune" at the same time or does it hinder power saving (My current setup is to use both....) ?

I could not find conclusive answers to this on the web. So better go to the source :-)

Thanks!

rickysarraf, 2018-02-10:
It is just a matter of choice. TLP and Laptop Mode Tools serve the same purpose. LMT has been around since, IIRC, 2005. TLP joined in later. But in terms of functionality, they may be the same. I've never used TLP myself.

I wouldn't suggest using `powertop` either, to do any tuning. I use LMT as the tuning tool and powertop (and others) as the reporting tool.

closed 2018-02-10 by rickysarraf

d-g, 2021-05-16:
FWIW, TLP seems to have at least one severe objective drawback over LMT: for a some unclear reason its documentation [1] is nonfree, and therefore should be normally missing from a distribution.

[1] https://linrunner.de/tlp/