rickysarraf/laptop-mode-tools #126: update manpages for user commands

rickysarraf, 2018-02-16:
```
rrs@priyasi:~/rrs-home/devel/Laptop-Mode-Tools/laptop-mode-tools (lmt-upstream)$ ./usr/sbin/laptop_mode --help
****************************
Following user commands are understood
status	    :	Display a Laptop Mode Tools power savings status
power-stats  :	Display the power statistics on the machine
power-events :	Trap power related events on the machine
help	    :	Display this help message (--help, -h)
version	    :	Display program version (--version, -v)
****************************
11:30 ♒♒♒   ☺    
```

mrdengbo, 2021-03-04:
I have updated the man pages of some commands #163 , and some descriptions may be inaccurate. Can you help check？

rickysarraf, 2021-03-04:
Closed with 25496b6

closed 2021-03-04 by rickysarraf