rickysarraf/laptop-mode-tools #127: dependency error?

GFdevelop, 2018-03-19:
Hello, I have this iussue (command not found):
`/usr/share/laptop-mode-tools/modules/hal-polling: riga 48: hal-disable-polling: comando non trovato`
I'm on arch linux, how to solve?

rickysarraf, 2018-03-20:
Device polling moved on to the newer kernel interface quite some time ago. Your kernel should be providing that interface.

What kernel are you running?
Please note that `hal-disable-polling` was deprecated/discontinued long time ago. The code there is for fallback and backwards compatibility only.

GFdevelop, 2018-03-20:
4.15.10-1-ARCH
I can't use hal-polling.conf? I have installed hal package from aur...is this package deprecated?

rickysarraf, 2018-03-21:
You can use the module. Just that the code should use the kernel sysfs interface instead of falling back and calling an obsolete/unsupported tool. I am not sure why that is not happening.

To dig further, would you be able to run the `hal-polling` module in debug mode ?
To do so, you need to enable `DEBUG=1` in `hal-polling.conf` and then restart laptop-mode-tools. Then the logs will be logged into syslog/journald.

GFdevelop, 2018-03-21:
where are logs? I have errors on boot, journalctl -p 4 -xb
```
mar 21 10:08:03 archlinux kernel: r8169 0000:04:00.1 enp4s0f1: rtl_ocp_gphy_cond == 1 (loop: 10, delay: 25).
mar 21 10:08:03 archlinux kernel: r8169 0000:04:00.1 enp4s0f1: rtl_ocp_gphy_cond == 1 (loop: 10, delay: 25).
mar 21 10:08:03 archlinux kernel: r8169 0000:04:00.1 enp4s0f1: rtl_ocp_gphy_cond == 1 (loop: 10, delay: 25).
mar 21 10:08:03 archlinux kernel: r8169 0000:04:00.1 enp4s0f1: rtl_ocp_gphy_cond == 1 (loop: 10, delay: 25).
mar 21 10:08:03 archlinux kernel: r8169 0000:04:00.1 enp4s0f1: rtl_ocp_gphy_cond == 1 (loop: 10, delay: 25).
mar 21 10:08:03 archlinux kernel: r8169 0000:04:00.1 enp4s0f1: rtl_ocp_gphy_cond == 1 (loop: 10, delay: 25).
```

(I'm using r8169aspm-dkms that enable aspm and blacklist default r8169)
and this:
`mar 21 10:08:37 archlinux bluetoothd[15526]: Failed to set mode: Blocked through rfkill (0x12)`

manually start laptop_mode:
```
sudo laptop_mode start
Enabling debug mode for module /etc/laptop-mode/conf.d/ethernet.conf
Enabling debug mode for module /etc/laptop-mode/conf.d/hal-polling.conf
Laptop mode
enabled, active
+ disableDebug /usr/share/laptop-mode-tools/modules/ethernet
+ . /usr/share/laptop-mode-tools/modules/ethernet
#! /bin/sh
#
# Laptop mode tools module: Ethernet power saving tweaks.
#

if [ x$CONTROL_ETHERNET = x1 ] || [ x$ENABLE_AUTO_MODULES = x1 -a x$CONTROL_ETHERNET = xauto ]; then

	# Let's check the binaries firts
	if [ -x /sbin/ethtool ]; then
		ETHTOOL=/sbin/ethtool
	elif [ -x /usr/sbin/ethtool ]; then
		ETHTOOL=/usr/sbin/ethtool
	else
		log "VERBOSE" "ethtool is not installed"
		ETHTOOL=/bin/false
	fi

	if [ -x /sbin/mii-tool ]; then
		MIITOOL=/sbin/mii-tool
	elif [ -x /usr/sbin/mii-tool ]; then
		MIITOOL=/usr/sbin/mii-tool
	else
		log "VERBOSE" "mii-tool is not installed"
		MIITOOL=/bin/false
	fi

	if [ -x /bin/ip ]; then
		IPTOOL=/bin/ip
	else
		log "VERBOSE" "ip is not installed"
		IPTOOL=/bin/false
	fi

	# Determine speed capability of physical device
	speed=`$MIITOOL -v $DEVICE 2>/dev/null | grep capabilities | tr ' ' '\n' |\
		sort -n | sed -ne '/^1.*/p' | cut -d "b" -f1`
	if [ -z "$speed" ]; then
		speed=0;
	fi
	max_s=0;
	min_s=100000;
	for s in $speed;
	do
		if [ $s -gt $max_s ]; then
			max_s=$s
		fi
		if [ $s -lt $min_s ]; then
[... 254 lines cut ...]
++ for DEV in $HAL_POLLING_DEVICES
+++ readlink /dev/sr0
++ basename /usr/share/laptop-mode-tools/modules/hal-polling
++ cut -d . -f1
++ tr '[:lower:]' '[:upper:]'
++ sed s/-/_/g
++ device=
++ '[' -f /sys/block//events_poll_msecs ']'
++ '[' -b /dev/sr0 ']'
++ hal-disable-polling --device /dev/sr0
/usr/share/laptop-mode-tools/modules/hal-polling: riga 48: hal-disable-polling: comando non trovato
++ log VERBOSE 'Executing command hal-disable-polling --device on device /dev/sr0'
++ '[' x1 = x1 ']'
++ '[' -x /usr/bin/logger -a VERBOSE '!=' STATUS ']'
++ '[' VERBOSE = MSG ']'
++ '[' VERBOSE = ERR ']'
++ '[' VERBOSE = VERBOSE ']'
++ '[' x0 = x1 ']'
++ '[' VERBOSE = VERBOSE ']'
++ '[' 1 = 0 ']'

+ '[' x1 = x1 ']'
+ set +vx
LCD Brightness Command succeeded.
```

rickysarraf, 2018-03-21:
Thanks. It seems you have a device, which is enumerated as `/dev/sr0`. And this device does not have a block sysfs interface. Hence we fallback to the old `HAL` way to disabling polling.
The only catch being your distribution may not be shipping `hal` any more, because as far as I know, it is obsolete/unsupported now.

I am still curious how you got `/dev/sr0`. Because the default configuration sets it at: `/dev/scd?`. Maybe a symlink. Hmmm.

rickysarraf, 2018-03-21:
Can you show me the output of `ls /sys/block/` ?

rickysarraf, 2018-03-22:
Okay. Thanks. So we have 2 things here.

1. Your device is enumerated as `/dev/sr0`, which is fine. Perhaps that interface does not have event polling exposed through sysfs. Can you also show the output of: `/sys/block/sr0/` ?

2. `hal-disable-polling` not being available is understood. HAL was deprecated long ago, and I doubt if there's any distribution shipping it today.

So, to summarize, there's little LMT can do here. Because either of the interfaces that we use is unavailable.

But for 1, there's still an issue. You mentioned that `/sys/block` did have `sr0`. But in the logs above, a `readlink /dev/sr0` failed. Which led to the fallback hal code. You'll have to dig a little deeper yourself because I do not have any machine with a cdrom drive.

What is the output of `readlink /dev/sr0` ?

GFdevelop, 2018-03-22:
ls -al /sys/block/sr0/
```
totale 0
drwxr-xr-x 8 root root    0 22 mar  2018 ./
drwxr-xr-x 3 root root    0 22 mar  2018 ../
-r--r--r-- 1 root root 4096 22 mar 10.54 alignment_offset
lrwxrwxrwx 1 root root    0 22 mar 10.54 bdi -> ../../../../../../../../virtual/bdi/11:0/
-r--r--r-- 1 root root 4096 22 mar 10.54 capability
-r--r--r-- 1 root root 4096 22 mar 10.54 dev
lrwxrwxrwx 1 root root    0 22 mar 10.48 device -> ../../../2:0:0:0/
-r--r--r-- 1 root root 4096 22 mar 10.54 discard_alignment
-r--r--r-- 1 root root 4096 22 mar 10.54 events
-r--r--r-- 1 root root 4096 22 mar 10.54 events_async
-rw-r--r-- 1 root root 4096 22 mar 10.54 events_poll_msecs
-r--r--r-- 1 root root 4096 22 mar 10.54 ext_range
drwxr-xr-x 2 root root    0 22 mar 10.48 holders/
-r--r--r-- 1 root root 4096 22 mar 10.54 inflight
drwxr-xr-x 2 root root    0 22 mar 10.52 integrity/
drwxr-xr-x 2 root root    0 22 mar 10.52 power/
drwxr-xr-x 3 root root    0 22 mar 10.49 queue/
-r--r--r-- 1 root root 4096 22 mar 10.54 range
-r--r--r-- 1 root root 4096 22 mar 10.54 removable
-r--r--r-- 1 root root 4096 22 mar 10.54 ro
-r--r--r-- 1 root root 4096 22 mar 10.54 size
drwxr-xr-x 2 root root    0 22 mar 10.52 slaves/
-r--r--r-- 1 root root 4096 22 mar 10.54 stat
lrwxrwxrwx 1 root root    0 22 mar 10.48 subsystem -> ../../../../../../../../../class/block/
drwxr-xr-x 2 root root    0 22 mar 10.52 trace/
-rw-r--r-- 1 root root 4096 22 mar 10.54 uevent
```

no output for readlink /dev/sr0

rickysarraf, 2018-03-23:
Thank you very much for this output. I can see the problem here with `readlink`

```
rrs@priyasi:/tmp$ readlink /tmp/c
b
11:42 ♒♒♒   ☺ 😄    
rrs@priyasi:/tmp$ readlink /tmp/b
11:42 ♒♒♒    ☹ 😟=> 1  
rrs@priyasi:/tmp$ file /tmp/b /tmp/c
/tmp/b: directory
/tmp/c: symbolic link to b
11:42 ♒♒♒   ☺ 😄    
```

The problem is that for a block device, `readlink` is returning nothing. Which is the cause of the problem in this case. Let me think what best can be done here.

closed 2018-03-23 by rickysarraf

rickysarraf, 2018-03-23:
@GFdevelop  Can you please try this fix ? https://github.com/rickysarraf/laptop-mode-tools/commit/53b7b0b0d4e671bf61cf8330c9f5fc3acb7ad829
It should fix your problem. Please report back your results.

rickysarraf, 2018-03-26:
Hello @GFdevelop . Any feedback ?

GFdevelop, 2018-03-26:
It don't give me hal polling error, hot to check if it work?

rickysarraf, 2018-03-26:
In the usual case, the cdrom drive will spin up every 30 seconds or a couple of minutes. I don't recall how often that was. With this setting, the automatic polling gets disabled.

GFdevelop, 2018-03-26:
ok I don't have any cdrom in this time, tonight I can do test with a cdrom. Sorry for the wait.

rickysarraf, 2018-04-08:
Any feedback @GFdevelop