rickysarraf/laptop-mode-tools #129: System hangs periodically

wkunkel, 2018-03-31:
Since updating to 1.72, I've been experiencing hangs periodically, where my system will quite rapidly slow down and then stop responding entirely. During these events, htop reports normal CPU and memory usage; although they seem to happen more frequently when doing CPU-intensive tasks like watching YouTube videos. Uninstalling laptop-mode-tools appears to have fixed the problem; at least, I haven't experienced a hang in two days when before they were every few hours, sometimes only minutes after rebooting.

I'm on a Lenovo ThinkPad T540p, using Arch Linux with the 4.15.13-1-ARCH kernel. I was running version 1.72.2 of laptop-mode-tools, although I had not updated to the previous releases of 1.72 so I cannot confirm whether they have the same issue.

rickysarraf, 2018-03-31:
This should be fixed with 1.72.2

I am currently traveling. I'll look into it once I'm back. Thanks

rickysarraf, 2018-03-31:
Your issue must be the same as #123
AHCI runtime_pm issues with ata disks on Linux 4.15+
Opened by rickysarraf 2 months ago
Bug  Not Our Bug

The analysis should apply to your use case too

wkunkel, 2018-03-31:
The issue seems related, but I'm not sure exactly what conclusions to draw from that thread. It looks like 1.72.2 should have stopped laptop-mode-tools from starting/stopping my disk, but the problem persists when using 1.72.2, and it looks like my disk is still being turned off. Plus, I would expect the freeze to coincide with the disk turning off and then failing to turn back on, but that's not what I see in dmesg:

[image]

rickysarraf, 2018-04-08:
Hello @wkunkel 

Does setting `CONTROL_AHCI_RUNTIME_PM=0` in `/etc/laptop-mode/conf.d/intel-sata-powermgmt.conf` solve your problem ?

amanoel, 2018-04-10:
Had the same issue, can confirm uninstalling laptop-mode-tools fixed it. Started to suspect it after noticing the system would hang after plugging/unplugging laptop, or right after coming back from suspend.

It wouldn't hang completely at first: I was still able to type things in the terminal, however no command would work anymore. I would them have to force shutdown my laptop. I was never able to find anything related to it in the system logs...

Right now I'm on an ASUS UX330U using Arch 4.15.14-1. I can try the proposed fix at some point, but right now I prefer having a stable system than trying things out.

rickysarraf, 2018-04-10:
I understand the issue is uncovered by Laptop Mode Tools. But the issue at hand is in the kernel. With 4.14, you shouldn't be seeing the problem. And nobody has spent time to bisect on what in the kernel has caused this problem.

rickysarraf, 2018-07-24:
Dupe of: https://github.com/rickysarraf/laptop-mode-tools/issues/123

closed 2018-07-24 by rickysarraf