rickysarraf/laptop-mode-tools #133: Cracking sound from mini-jack

F171pp0, 2018-10-01:
I have `laptop-mode-tools` 1.72.2-3 from `[Arch's AUR](https://aur.archlinux.org/packages/laptop-mode-tools/)`and if I connect my mini-jack earphones I hear a noisy crackling sound. If I play some music or any audio file the crackling sound disappears, but when the music is stopped after few seconds it reappears. Already tried to disable `intel-hda-powersave` and `ac97-powersave` modules, but I couldn't get rid of this annoying sound. Only entirely removing laptop-mode-tools I can use my laptop with earphones.

rickysarraf, 2018-10-01:
You mean there is a continuous noise, that is heard a couple of seconds after the music stops playing ? If it is related to the sound card, it should have stopped after you disabled the respective modules.

Can you share the journal logs for `laptop-mode-tools` ? Also some more details about your hardware ?
I've not had any such issues nor have I had any such reports yet.

rickysarraf, 2018-10-02:
No. this is not the output I'm looking for. Please enable verbose logging in `/etc/laptop-mode/laptop-mode.conf` by setting `VERBOSE_OUTPUT=1`. Then restart `laptop-mode.service`.

Then, share the output of: `journalctl --unit laptop-mode.service`

F171pp0, 2018-10-02:
Sorry, here is the [output](https://pastebin.com/LEbAVLKD)

rickysarraf, 2018-10-03:
```
ott 02 14:40:18 mi-arch laptop-mode[11648]: Invoking module /usr/share/laptop-mode-tools/modules/intel-hda-powersave.
ott 02 14:40:18 mi-arch laptop_mode[11335]: Invoking module /usr/share/laptop-mode-tools/modules/intel-hda-powersave.
ott 02 14:40:18 mi-arch laptop-mode[11662]: Executing comand
ott 02 14:40:18 mi-arch laptop_mode[11335]: Executing comand
ott 02 14:40:18 mi-arch laptop_mode[11335]: Intel HDA power save mode enabled with timeout 2.
ott 02 14:40:18 mi-arch laptop_mode[11335]: Intel HDA Controller power save set to Y.
```

So you seem to have an Intel HDA sound card, whcih is a very common one in use today. I'm using the same and haven't run into any problem.

You mentioned initially that you disabled this module. But from the logs, it looks like this module is still invoked. Can you please check and report back the actual state ?

Also, check to see if disabling the `runtime-pm` module solves your problem ? As far as I see, there's nothing wrong with laptop-mode-tools here. It seems like your device may not be playing well with power saving mode. We definitely need to narrow down what in `laptop-mode-tools` is triggering the power saving for your sound card. At this moment, you have 2 suspects: `intel-hda-powersave` and `runtime-pm`

F171pp0, 2018-10-04:
Yes, you are right, it was enabled because I had reinstalled it, but as you can see [here](https://pastebin.com/ebHs1L3Z) it is now disabled and I continue to hear the noise. 
Disabling `runtime-pm` solves the problem, thank you very much.

F171pp0, 2018-10-04:
Yes, you are right, it was enabled because I had reinstalled it, but as you can see [here](https://pastebin.com/ebHs1L3Z) it is now disabled and I continue to hear the noise. 
--Update--
Disabling `runtime-pm` solves the problem, thank you very much. Now I can also reenable `intel-hda-powersave` without any annoying sound

rickysarraf, 2018-10-05:
I'm glad you were able to work around it. But remember, disabling `runtime-pm` comes at the cost of additional power consumption. You may want to check under `/sys/bus/pci/devices/` on which devices power saving is causing this problem, and only disable that device in `runtime-pm.conf`. It may be a repetitive task to determine the actual device, but if you can, you would get much better battery backup.

As for this bug report, I am closing it. Thanks.

closed 2018-10-05 by rickysarraf