rickysarraf/laptop-mode-tools #135: How do I enable laptop-mode-tools?

davismcc, 2018-11-27:
Hi there,

I am running Ubuntu 18.04 LPS on a Dell XPS 13 (2018) laptop. I have installed laptop-mode-tools with `apt`, but `cat /proc/sys/vm/laptop_mode` returns 0, even after multiple reboots.

How do I enable laptop-mode-tools?

Apologies if I missed it in the README or wiki, but I did not read it anywhere. Maybe it seems blindingly obvious how to do this, but would be great to spell it out in the docs for dullards like me.

Thanks and best wishes,
Davis

davismcc, 2018-11-28:
OK, in the meantime it seems to have fixed itself. Perhaps just some time off AC power prompted laptop-mode-tools to enable itself? In any case, an extra explanatory sentence or two in the FAQ would be helpful for neophytes like me.

closed 2018-11-28 by davismcc

rickysarraf, 2018-11-29:
Hi. I use the same hardware with `laptop-mode-tools` installed. :-)

I agree the status could be better. There have been some enhancements about it.

First, with systems running `systemd`, you could get the status from there:

```
rrs@priyasi:~/bin$ systemctl status laptop-mode
● laptop-mode.service - Laptop Mode Tools
   Loaded: loaded (/lib/systemd/system/laptop-mode.service; enabled; vendor preset: enabled)
   Active: active (exited) since Tue 2018-11-27 20:57:11 IST; 1 day 15h ago
     Docs: man:laptop_mode(8)
           man:laptop-mode.conf(8)
           http://github.com/rickysarraf/laptop-mode-tools
  Process: 9891 ExecReload=/usr/sbin/laptop_mode auto (code=exited, status=0/SUCCESS)
 Main PID: 1329 (code=exited, status=0/SUCCESS)

Nov 29 12:30:03 priyasi laptop_mode[8468]: enabled, not active [unchanged]
Nov 29 12:30:03 priyasi systemd[1]: Reloaded Laptop Mode Tools.
Nov 29 12:32:34 priyasi systemd[1]: Reloading Laptop Mode Tools.
Nov 29 12:32:34 priyasi laptop_mode[9033]: Laptop mode
Nov 29 12:32:34 priyasi laptop_mode[9033]: enabled, not active [unchanged]
Nov 29 12:32:34 priyasi systemd[1]: Reloaded Laptop Mode Tools.
Nov 29 12:35:10 priyasi systemd[1]: Reloading Laptop Mode Tools.
Nov 29 12:35:10 priyasi laptop_mode[9891]: Laptop mode
Nov 29 12:35:10 priyasi laptop_mode[9891]: enabled, not active [unchanged]
Nov 29 12:35:10 priyasi systemd[1]: Reloaded Laptop Mode Tools.
12:35 ♒♒♒   ☺ 😄    
```

And in the next release of `laptop-mode-tools`, you will also have some reporting available from the tool itself.

```
rrs@priyasi:~/bin$ sudo laptop_mode -h
****************************
Following user commands are understood
status      :   Display a Laptop Mode Tools power savings status
power-stats  :  Display the power statistics on the machine
power-events :  Trap power related events on the machine
help        :   Display this help message (--help, -h)
version     :   Display program version (--version, -v)
****************************
12:34 ♒♒♒   ☺ 😄    
rrs@priyasi:~/bin$ sudo laptop_mode status
Mounts:
   /dev/mapper/nvme0n1p4_crypt on / type btrfs (rw,noatime,compress=zstd,ssd,space_cache,autodefrag,subvolid=5,subvol=/)
   /dev/nvme0n1p3 on /boot type ext4 (rw,relatime)
   /dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
   /dev/fuse on /run/user/1000/doc type fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
 
Drive power status:
   Cannot read /dev/[hs]d[abcdefgh], permission denied - /usr/sbin/laptop_mode needs to be run as root
 
(NOTE: drive settings affected by Laptop Mode cannot be retrieved.)
 
Readahead states:
   /dev/mapper/nvme0n1p4_crypt: 128 kB
   /dev/nvme0n1p3: 128 kB
   /dev/nvme0n1p1: 128 kB
 
Laptop Mode Tools is allowed to run: /var/run/laptop-mode-tools/enabled exists.
 
/proc/sys/vm/laptop_mode:
   0
 
/proc/sys/vm/dirty_ratio:
   40
 
/proc/sys/vm/dirty_background_ratio:
   10
 
/proc/sys/vm/dirty_expire_centisecs:
   3000
 
/proc/sys/vm/dirty_writeback_centisecs:
   500
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq:
   2001000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_cur_freq:
   400000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu1/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_cur_freq:
   400000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu2/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_cur_freq:
   400000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu3/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu4/cpufreq/cpuinfo_cur_freq:
   400000
 
/sys/devices/system/cpu/cpu4/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu4/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu5/cpufreq/cpuinfo_cur_freq:
   400000
 
/sys/devices/system/cpu/cpu5/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu5/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu6/cpufreq/cpuinfo_cur_freq:
   400000
 
/sys/devices/system/cpu/cpu6/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu6/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu7/cpufreq/cpuinfo_cur_freq:
   400000
 
/sys/devices/system/cpu/cpu7/cpufreq/cpuinfo_max_freq:
   2001000
 
/sys/devices/system/cpu/cpu7/cpufreq/cpuinfo_min_freq:
   400000
 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu4/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu5/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu6/cpufreq/scaling_governor:
   schedutil
 
/sys/devices/system/cpu/cpu7/cpufreq/scaling_governor:
   schedutil
 
/proc/acpi/button/lid/LID0/state:
   state:      open
 
/sys/class/power_supply/AC/online:
   1
 
/sys/class/power_supply/BAT0/status:
   Charging
 
12:34 ♒♒♒   ☺ 😄    
```