rickysarraf/laptop-mode-tools #137: Udev rule 99-laptop-mode.rules slows down boot time

laserbat, 2019-01-11:
On my system, with 99-laptop-mode.rules overriden (`touch /etc/udev/rules.d/99-laptop-mode.rules`) it takes roughly 0.5  for udev to settle devices. Without this file, it takes around **6** seconds.

I used [this script](https://gist.github.com/laserbat/5a7d0978ca79f9369c4584c0632cce1d) to precisely measure time it takes for udev to settle devices.

[Here's](https://gist.github.com/laserbat/eb083bfc76820b20947b01e192846af0) gist with two outputs, with and without the override.

I used [execsnoop from bcc-tools](https://github.com/iovisor/bcc) to [look at what exactly udev is doing](https://gist.github.com/laserbat/5b18afec37e0953ad9362e55a0af5e81), and it calls /usr/lib/udev/lmt-udev multiple times. Since udev waits for a process to finish, this creates noticeable lag.

rickysarraf, 2019-01-11:
What do you mean by `udev to settle devices` ?
AFAIK, `udev` parallelizes its rules processing.

laptop-mode-tools is invoked early, at boot as you see. And those numbers are correct. laptop-mode-tools takes around 5 seconds to execute. There can be multiple invocations but only one instance runs at any given time.

But none of this should block anything and make it noticeable. 

Here's what it looks like on my machine:

```
rrs@priyasi:/var/tmp/Debian-Build/temp/ $ systemd-analyze blame
          5.098s logrotate.service
          2.833s networking.service
          1.274s tor@default.service
          1.053s systemd-tmpfiles-clean.service
           503ms dev-mapper-nvme0n1p4_crypt.device
           327ms systemd-resolved.service
           318ms NetworkManager-wait-online.service
           318ms snapper-boot.service
           285ms systemd-timesyncd.service
           284ms ModemManager.service
           252ms exim4.service
           235ms systemd-journal-flush.service
           230ms upower.service
           189ms systemd-logind.service
           183ms laptop-mode.service
           172ms NetworkManager.service
           160ms zramswap.service
           160ms wpa_supplicant.service
           149ms lvm2-lvmetad.service
           144ms geoclue.service
           133ms systemd-networkd.service
           118ms lm-sensors.service
           117ms systemd-fsck@dev-disk-by\x2duuid-EACF\x2dBB61.service
           117ms systemd-fsck@dev-disk-by\x2duuid-6ba243d1\x2d06f8\x2d4440\x2d9541\x2de324649292bb.service
           112ms bluetooth.service
```

laserbat, 2019-01-11:
> What do you mean by udev to settle devices ?

I'm talking about this sequence of commands that [are executed on my system as a part of startup](https://github.com/void-linux/void-runit/blob/master/core-services/02-udev.sh):

```
    ${_udevd} --daemon
    udevadm trigger --action=add --type=subsystems
    udevadm trigger --action=add --type=devices
    udevadm settle
```

I don't have systemd on my system (using runit on Void Linux). 

> But none of this should block anything and make it noticeable.

From `man udev` and this issue on stackoverflow: https://askubuntu.com/questions/667922/udev-script-doesnt-run-in-the-background

```
           This can only be used for very short-running foreground tasks.
           Running an event process for a long period of time may block all
           further events for this or a dependent device.

           Starting daemons or other long-running processes is not appropriate
           for udev; the forked processes, detached or not, will be
           unconditionally killed after the event handling has finished.
```

Running `sudo time -v ./lmt-udev auto force` produces [this output](https://gist.github.com/laserbat/c33d57cb521e19a5d023d8aaf1d76eb9).

laserbat, 2019-01-11:
So, my guess is that some blocking does happen. Either as a part of laptop-mode rules, or from other rules dependent on devices that are managed by laptop-mode. I'm sorry if I'm mistaken, I'm pretty new to analyzing udev rules/performance

rickysarraf, 2019-01-11:
> So, my guess is that some blocking does happen. Either as a part of laptop-mode rules, or from other rules dependent on devices that are managed by laptop-mode. I'm sorry if I'm mistaken, I'm pretty new to analyzing udev rules/performance

laptop-mode-tools does not manage any device. It just enquires them. 

For `udev`, laptop-mode-tools shouldn't be blocking anything for parallel execution. But yes, for udev to exit clean and move ahead, it does need for the laptop-mode-tools' scripts to complete, which on an average take around 5 seconds. This execution time is something I have already optimized (parallel module execution etc) but if you see further enhancements, please do share.

rickysarraf, 2019-01-11:
> I'm talking about this sequence of commands that [are executed on my system as a part of startup](https://github.com/void-linux/void-runit/blob/master/core-services/02-udev.sh):
>
> ```
>     ${_udevd} [...]

This is correct. And that is why the script invocation through udev rules takes extra care to ensure it can completely run.

laserbat, 2019-01-11:
> For udev, laptop-mode-tools shouldn't be blocking anything for parallel execution. But yes, for udev to exit clean and move ahead, it does need for the laptop-mode-tools' scripts to complete, which [...]

So, if I understand correctly, "udevadm settle" will keep waiting until laptop-mode setup is finished and hence cause delay?

rickysarraf, 2019-01-11:
> So, if I understand correctly, "udevadm settle" will keep waiting until laptop-mode setup is finished and hence cause delay?

Things may have changed lately but I'll share my understanding of how udev behaves.

Here's what the official documentation says:
```
udevadm [--help] [--version] [--debug] COMMAND [COMMAND OPTIONS]

Send control commands or test the device manager.

Commands:
  info          Query sysfs or the udev database
  trigger       Request events from the kernel
  settle        Wait for pending udev events
  control       Control the udev daemon
```
So, by this, it is correct for udev to wait until everything has settled.

BUT, under normal scenarios, you don't need to run `udevadm settle`. That is only needed where you have something in your boot chain that depends on something remote. A good example is a **Remote Block Device**, which can take quite some time to show up on the host machine. In such scenarios, you run `settle` for the system to stabilize and then act on it.
 
> If that's the case, I'll close the issue and instead take a look at what can be done with void boot system to speed up the process.

Do you have a good reason to run the `settle` command ?

laserbat, 2019-01-11:
> Do you have a good reason to run the settle command ?

I don't know the precise reasoning behind it, but commenting it out makes some (non-critical) parts of my system to break. Namely, pulseaudio stopped working  and `rfkill unblock wlan` in /etc/rc.local doesn't seem to function correctly.

Now, both of these issues aren't hard to fix, but it does seem to me that there is at least some reasoning behind using udevadm settle in startup script.

rickysarraf, 2019-01-11:
> > Do you have a good reason to run the settle command ?
> 
> I don't know the precise reasoning behind it, but commenting it out makes some (non-critical) parts of my system to break. Namely, pulseaudio stopped working and `rfkill unblock wlan` in /etc/rc.local doesn't seem to function correctly.

It looks like a case of an `init` manager which does not have the dependencies well defined.
I have never used `runit` but I doubt that any init manager wouldn't have dependency tracking. So the bug must be in the init scripts.

> 
> Now, both of these issues aren't hard to fix, but it does seem to me that there is at least some reasoning behind using udevadm settle in startup script.

Yes. There is. But that mostly ends up in `initrd`.  Because, usually, that is where you open your `rootfs` block device and other stuff. Or discover your network interface, if you are fetching a `root` block device over the network, or your `rootfs` file system over the network. Even for local disks, a `settle` should be used in `initrd`. Like, for example, when you have your `rootfs` over an encrypted LUKS block device, whiich itself is on a local/remote physical block device.

Here's a quick check, on Debian:

```
rrs@priyasi:/usr/share/initramfs-tools$ grep -Ri settle *
scripts/init-top/udev:udevadm settle || true
scripts/functions:      udevadm settle ${1:+--timeout=$1}
15:26 ♒♒♒   ☺ 😄    

rrs@priyasi:/usr/share/initramfs-tools$ cat scripts/init-top/udev 
#!/bin/sh -e

PREREQS=""

prereqs() { echo "$PREREQS"; }

case "$1" in
    prereqs)
    prereqs
    exit 0
    ;;
esac

if [ -w /sys/kernel/uevent_helper ]; then
        echo > /sys/kernel/uevent_helper
fi

if [ "$quiet" = "y" ]; then
        log_level=notice
else
        log_level=info
fi

SYSTEMD_LOG_LEVEL=$log_level /lib/systemd/systemd-udevd --daemon --resolve-names=never

udevadm trigger --type=subsystems --action=add
udevadm trigger --type=devices --action=add
udevadm settle || true

# Leave udev running to process events that come in out-of-band (like USB
# connections)
15:26 ♒♒♒   ☺ 😄    

```

This script, which is run from `initrd`, is self explanatory.

closed 2019-01-11 by laserbat