rickysarraf/laptop-mode-tools #140: Attaching USB input devices is slow when laptop-mode-tools is installed (laptop_mode not running)

vozhyk-, 2019-02-16:
When I close then open the lid in my laptop (I see the Wacom "Pen and multitouch sensor" USB device disconnecting and connecting in the kernel logs then), the touchscreen does not respond for 1 to 5 seconds. When I disconnect then connect a USB trackball, the cursor does not move for a similar period of time. Starting the `laptop_mode` service is disabled.

If I uninstall `laptop-mode-tools`, there is no noticeable delay when reconnecting those devices.

I've found that there is `/lib/udev/rules.d/99-laptop-mode.rules` that runs `lmt-udev force` when a USB device is added, and I see #137 mentioning a similar slowdown, but on boot on a different system.

- Laptop: Thinkpad X1 Yoga 3rd Gen.
- Distribution: Gentoo Linux, without `systemd`.
- `laptop-mode-tools` version: `app-laptop/laptop-mode-tools-1.72.2-r1`.
- `eudev` version: `sys-fs/eudev-3.2.7-r1`.

rickysarraf, 2019-02-18:
THis is working as designed. Laptop Mode Tools applies aggressive power savings by default. You can blacklist the relevant usb device in `/etc/laptop-mode/conf.d/runtime-pm.conf` and then power savings will not be applied for that device.

PF4Public, 2019-07-20:
> You can blacklist the relevant usb device in /etc/laptop-mode/conf.d/runtime-pm.conf and then power savings will not be applied for that device.

I have the same issue with a mouse and LMT.
I have tried adding mouse ID and the USB hub, it is connected to, to the blacklist to no avail.
Also tried adding hid-generic and hid-raw to blacklist. This also didn't change anything.

What do we actually need to write into `/etc/laptop-mode/conf.d/runtime-pm.conf` to have a device blacklisted? It has already `hub` and `usbhid` among others, but this seems to be effectless.

rickysarraf, 2019-07-24:
@PF4Public I put those defaults keeping in mind that pluggable USB devices are something people would want to not have power savings enabled for.

I just checked here locally to verify that is working or not. But, for me, a plugged-in usb mouse is working fine.

This is what I have in `runtime-pm.conf`
```
# The list of Device IDs that should not use autosuspend. Use system commands or
# look into sysfs to find out the IDs of your devices.
# Example: AUTOSUSPEND_RUNTIME_DEVID_BLACKLIST="046d:c025 0123:abcd"
AUTOSUSPEND_RUNTIME_DEVID_BLACKLIST="18d1:d00d ffff:5678 05c6:f00e"

# The list of 'device driver types' or 'device types' that should not use autosuspend.
# The driver type is given by "DRIVER=..." in a device's uevent file.
# The driver type is given by "DEVTYPE=..." in a device's uevent file.
# Example: AUTOSUSPEND_RUNTIME_DEVTYPE_BLACKLIST="usbhid usb-storage"
AUTOSUSPEND_RUNTIME_DEVTYPE_BLACKLIST="hub usbhid usb-storage"
```

And my external USB mouse is:
```
rrs@chutzpah:/sys/class$ sudo lsusb 
[sudo] password for rrs: 
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 016: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 002 Device 003: ID 04ca:7032 Lite-On Technology Corp. HP HD Webcam
Bus 002 Device 015: ID 05c6:f00e Qualcomm, Inc. Mobile Router
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
14:32 ♒♒♒   ☺       
```

As you can see, I haven't blacklisted the mouse's device id. Because the generic `usbhid` blacklist is enough. 

You can also tally the below reporting from `powertop`. It is inline with what I have set in my `runtime-pm.conf` file.
```
>> Bad           Wireless Power Saving for interface wlan0                                                              
   Good          NMI watchdog should be turned off
   Bad           VM writeback timeout
   Bad           Enable SATA link power management for host3
   Bad           Enable SATA link power management for host1
   Bad           Enable SATA link power management for host2
   Bad           Enable SATA link power management for host0
   Bad           Enable Audio codec power management
   Good          Autosuspend for USB device HP HD Webcam [DDZLE019I7MK1R]
   Good          Autosuspend for USB device xHCI Host Controller [usb2]
   Bad           Autosuspend for unknown USB device 1-1 (8087:8000)
   Bad           Autosuspend for USB device USB Optical Mouse [PixArt]   
   Good          Autosuspend for USB device EHCI Host Controller [usb1]
   Good          Autosuspend for USB device xHCI Host Controller [usb3]
   Good          Runtime PM for I2C Adapter i2c-3 (i915 gmbus dpc)
   Good          Runtime PM for I2C Adapter i2c-1 (i915 gmbus vga)
   Good          Runtime PM for I2C Adapter i2c-4 (i915 gmbus dpb)
   Good          Runtime PM for I2C Adapter i2c-2 (i915 gmbus panel)
   Good          Runtime PM for I2C Adapter i2c-0 (i915 gmbus ssc)
   Good          Runtime PM for I2C Adapter i2c-5 (i915 gmbus dpd)
   Bad           Autosuspend for USB device Mobile Router [Qualcomm]
   Good          Runtime PM for PCI Device Intel Corporation 8 Series PCI Express Root Port 1
   Good          Runtime PM for PCI Device Intel Corporation 8 Series PCI Express Root Port 2
   Good          Runtime PM for PCI Device Broadcom Inc. and subsidiaries BCM43228 802.11a/b/g/n
   Good          Runtime PM for PCI Device Intel Corporation 8 Series LPC Controller
   Good          Runtime PM for PCI Device Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
   Good          Runtime PM for PCI Device Intel Corporation 8 Series HD Audio Controller
   Good          Runtime PM for PCI Device Intel Corporation 8 Series HECI #0
   Good          Runtime PM for PCI Device Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader
   Good          Runtime PM for PCI Device Intel Corporation 8 Series PCI Express Root Port 4
   Good          Runtime PM for PCI Device Intel Corporation 8 Series USB xHCI HC
```

It is unfortunate that the same is not working for you. Can you try out the things I have demoed above?
Ultimately, if you have a generic class device, you can blacklist it by its device type. Otherwise, if it is too confusing, you can always explicitly disable the device by its `device ID`

matthijskooijman, 2020-02-26:
I was also running into this with serial devices (Arduinos) that take a few seconds to get their udev rules applied, which sometimes causes firmware uploads to fail due to permissions not being set yet.

I believe the problem is not so much that laptop-mode powersaving is applied for these devices, but that whenever *any* USB device is inserted, `laptop_mode force` is run synchronously from udev which re-applies the laptopt-mode state, but also blocks udev for a few seconds, delaying further processing of the USB device. This probably means that blacklisting will not help, since the full lmt state is reapplied (it will just skip this particular device, but still take some time).

I wonder:
 - Does lmt really need to force apply everything again? If it could maybe *just* apply things to the new USB device, it might be faster?
 - Does lmt really need to run synchronously from udev? If it would spawn into the background, udev could continue processing right away.

matthijskooijman, 2020-02-26:
> Does lmt really need to force apply everything again? If it could maybe just apply things to the new USB device, it might be faster?

It actually seems there is already a (commented) rule in the udev rules file that seems to do just this:

    ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force modules=runtime-pm devices=%k"

I just tried this, and that seems to be fast enough to make udev process the USB device a lot faster (250-300ms, vs 60-150ms when the udev rule is disabled completely, and just under 3s with the default rule). I measured these times by looking at `udevadm monitor`, looking at the difference between the first event (add) and last event (bind) for the USB device.

rickysarraf, 2020-02-26:
> I was also running into this with serial devices (Arduinos) that take a few seconds to get their udev rules applied, which sometimes causes firmware uploads to fail due to permissions not being set [...]

laptop-mode rules apply almost at the end in the udev rules processing chain.

When you say "delaying further processing of the USB device", do you mean the udev rules ? Because, given we run quite late in the chain, that shouldn't be the effect.

> I wonder:
> 
>     * Does lmt really need to force apply everything again? If it could maybe _just_ apply things to the new USB device, it might be faster?
>

I'll have to dig the git history and the bug report. But yes, there were reasons to make this the default mode. I left the other option commented out for this very reason. If there are users that need further fine grained control, they could uncomment and use it. 

>     * Does lmt really need to run synchronously from udev? If it would spawn into the background, udev could continue processing right away.

matthijskooijman, 2020-02-26:
> When you say "delaying further processing of the USB device", do you mean the udev rules ? Because, given we run quite late in the chain, that shouldn't be the effect.

In practice, I see that applying the `chgrp dialout` to my `/dev/ttyACM0` serial device is delayed a few seconds, which is done using a udev rule (`/lib/udev/rules.d/50-udev-default.rules`).

What I suspect is that the add event for the USB device itself delays processing of the USB interfaces/subdevices, such as the serial device. Here's what `usbadm monitor` gives when plugging in with the default rules:
```
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[70089.912809] add      /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2 (usb)
KERNEL[70089.912929] add      /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0 (usb)
KERNEL[70089.913322] add      /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/tty/ttyACM0 (tty)
KERNEL[70089.913496] bind     /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0 (usb)
KERNEL[70089.913626] add      /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.1 (usb)
KERNEL[70089.913723] bind     /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.1 (usb)
KERNEL[70089.913808] add      /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2 (usb)
KERNEL[70089.913903] bind     /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2 (usb)
KERNEL[70090.165970] add      /kernel/slab/:A-0000192/cgroup/cred_jar(18217:session-c93.scope) (cgroup)
KERNEL[70090.166032] add      /kernel/slab/:A-0000256/cgroup/filp(18217:session-c93.scope) (cgroup)
KERNEL[70090.166254] add      /kernel/slab/:A-0000128/cgroup/pid(18217:session-c93.scope) (cgroup)
KERNEL[70090.166287] add      /kernel/slab/:A-0000072/cgroup/eventpoll_pwq(18217:session-c93.scope) (cgroup)
KERNEL[70090.166315] add      /kernel/slab/sock_inode_cache/cgroup/sock_inode_cache(18217:session-c93.scope) (cgroup)
KERNEL[70090.166339] add      /kernel/slab/:A-0001024/cgroup/UNIX(18217:session-c93.scope) (cgroup)
KERNEL[70090.166363] add      /kernel/slab/dentry/cgroup/dentry(18217:session-c93.scope) (cgroup)
UDEV  [70090.169987] add      /kernel/slab/:A-0001024/cgroup/UNIX(18217:session-c93.scope) (cgroup)
KERNEL[70090.170269] add      /kernel/slab/skbuff_head_cache/cgroup/skbuff_head_cache(18217:session-c93.scope) (cgroup)
KERNEL[70090.170502] add      /kernel/slab/kmalloc-512/cgroup/kmalloc-512(18217:session-c93.scope) (cgroup)
UDEV  [70090.171731] add      /kernel/slab/:A-0000072/cgroup/eventpoll_pwq(18217:session-c93.scope) (cgroup)
KERNEL[70090.172004] add      /kernel/slab/kmalloc-64/cgroup/kmalloc-64(18217:session-c93.scope) (cgroup)
UDEV  [70090.172614] add      /kernel/slab/:A-0000192/cgroup/cred_jar(18217:session-c93.scope) (cgroup)
UDEV  [70090.173959] add      /kernel/slab/sock_inode_cache/cgroup/sock_inode_cache(18217:session-c93.scope) (cgroup)
UDEV  [70090.174501] add      /kernel/slab/:A-0000256/cgroup/filp(18217:session-c93.scope) (cgroup)
UDEV  [70090.175312] add      /kernel/slab/:A-0000128/cgroup/pid(18217:session-c93.scope) (cgroup)
UDEV  [70090.177830] add      /kernel/slab/dentry/cgroup/dentry(18217:session-c93.scope) (cgroup)
KERNEL[70090.179962] add      /kernel/slab/task_struct/cgroup/task_struct(18217:session-c93.scope) (cgroup)
KERNEL[70090.180002] add      /kernel/slab/:A-0000080/cgroup/task_delay_info(18217:session-c93.scope) (cgroup)
KERNEL[70090.180026] add      /kernel/slab/:A-0000704/cgroup/files_cache(18217:session-c93.scope) (cgroup)
KERNEL[70090.180048] add      /kernel/slab/:A-0000064/cgroup/anon_vma_chain(18217:session-c93.scope) (cgroup)
KERNEL[70090.180071] add      /kernel/slab/sighand_cache/cgroup/sighand_cache(18217:session-c93.scope) (cgroup)
KERNEL[70090.180092] add      /kernel/slab/:A-0001088/cgroup/signal_cache(18217:session-c93.scope) (cgroup)
KERNEL[70090.180112] add      /kernel/slab/mm_struct/cgroup/mm_struct(18217:session-c93.scope) (cgroup)
KERNEL[70090.180132] add      /kernel/slab/:A-0000208/cgroup/vm_area_struct(18217:session-c93.scope) (cgroup)
KERNEL[70090.180150] add      /kernel/slab/anon_vma/cgroup/anon_vma(18217:session-c93.scope) (cgroup)
UDEV  [70090.182889] add      /kernel/slab/:A-0000064/cgroup/anon_vma_chain(18217:session-c93.scope) (cgroup)
UDEV  [70090.184291] add      /kernel/slab/:A-0001088/cgroup/signal_cache(18217:session-c93.scope) (cgroup)
UDEV  [70090.186420] add      /kernel/slab/task_struct/cgroup/task_struct(18217:session-c93.scope) (cgroup)
UDEV  [70090.186671] add      /kernel/slab/skbuff_head_cache/cgroup/skbuff_head_cache(18217:session-c93.scope) (cgroup)
UDEV  [70090.187160] add      /kernel/slab/sighand_cache/cgroup/sighand_cache(18217:session-c93.scope) (cgroup)
UDEV  [70090.189724] add      /kernel/slab/mm_struct/cgroup/mm_struct(18217:session-c93.scope) (cgroup)
UDEV  [70090.190053] add      /kernel/slab/kmalloc-512/cgroup/kmalloc-512(18217:session-c93.scope) (cgroup)
UDEV  [70090.190610] add      /kernel/slab/:A-0000704/cgroup/files_cache(18217:session-c93.scope) (cgroup)
UDEV  [70090.193474] add      /kernel/slab/:A-0000080/cgroup/task_delay_info(18217:session-c93.scope) (cgroup)
UDEV  [70090.193533] add      /kernel/slab/kmalloc-64/cgroup/kmalloc-64(18217:session-c93.scope) (cgroup)
UDEV  [70090.197027] add      /kernel/slab/:A-0000208/cgroup/vm_area_struct(18217:session-c93.scope) (cgroup)
[... cut ...]
UDEV  [70092.393660] bind     /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0 (usb)
UDEV  [70092.399722] bind     /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2 (usb)
```

All these kernel/cgroup lines seem to be generated by laptop-mode-tools (they are absent when I remove the lmt rule from udev). As you can see, at least some part of the ttyACM event is listed *after* all this cgroup stuff, so I suspect that the dialout rule happens there (though there is also a ttyACM event *before* the lmt stuff, so I'm not entirely sure how to interpret this output).

rickysarraf, 2020-02-26:
I did a plug/unplug of my USB Type-C headphone and I don't get similar results.

```
rrs@priyasi:~$ udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[104172.404179] change   /devices/platform/USBC000:00/typec/port0 (typec)
KERNEL[104172.404202] add      /devices/platform/USBC000:00/typec/port0/port0-partner (typec)
UDEV  [104172.405772] change   /devices/platform/USBC000:00/typec/port0 (typec)
UDEV  [104172.406712] add      /devices/platform/USBC000:00/typec/port0/port0-partner (typec)
KERNEL[104173.345985] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/wakeup/wakeup45 (wakeup)
KERNEL[104173.346069] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/wakeup/wakeup46 (wakeup)
KERNEL[104173.346332] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0 (pci)
KERNEL[104173.346441] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/pci_bus/0000:04 (pci_bus)
UDEV  [104173.350720] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/wakeup/wakeup45 (wakeup)
KERNEL[104173.474460] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:26/wakeup/wakeup47 (wakeup)
KERNEL[104173.476850] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0 (pci)
KERNEL[104173.476927] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:27/wakeup/wakeup48 (wakeup)
UDEV  [104173.476986] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/wakeup/wakeup46 (wakeup)
KERNEL[104173.477057] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0 (pci)
KERNEL[104173.477097] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/wakeup/wakeup49 (wakeup)
KERNEL[104173.477162] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0 (pci)
KERNEL[104173.477226] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0 (pci)
KERNEL[104173.477264] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/pci_bus/0000:05 (pci_bus)
KERNEL[104173.477299] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0/pci_bus/0000:06 (pci_bus)
KERNEL[104173.477333] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/pci_bus/0000:39 (pci_bus)
KERNEL[104173.486911] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/device:29/wakeup/wakeup50 (wakeup)
UDEV  [104173.487050] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0 (pci)
KERNEL[104173.487100] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0 (pci)
UDEV  [104173.487130] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/device:29/wakeup/wakeup50 (wakeup)
KERNEL[104173.487156] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0/pci_bus/0000:3a (pci_bus)
KERNEL[104173.487196] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0 (pci)
KERNEL[104173.487222] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:04:00.0:pcie210 (pci_express)
KERNEL[104173.487262] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0 (pci)
KERNEL[104173.487286] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0/0000:04:01.0:pcie210 (pci_express)
KERNEL[104173.487326] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0 (pci)
KERNEL[104173.487349] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:04:02.0:pcie210 (pci_express)
KERNEL[104173.487386] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0 (pci)
KERNEL[104173.487411] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0/0000:04:04.0:pcie210 (pci_express)
KERNEL[104173.487447] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0 (pci)
KERNEL[104173.487556] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3 (usb)
KERNEL[104173.487656] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-0:1.0 (usb)
UDEV  [104173.487819] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/wakeup/wakeup49 (wakeup)
UDEV  [104173.487872] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:27/wakeup/wakeup48 (wakeup)
KERNEL[104173.488250] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-0:1.0 (usb)
KERNEL[104173.488333] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3 (usb)
KERNEL[104173.488379] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/wakeup/wakeup51 (wakeup)
KERNEL[104173.489673] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4 (usb)
UDEV  [104173.489742] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:26/wakeup/wakeup47 (wakeup)
KERNEL[104173.489805] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4/4-0:1.0 (usb)
KERNEL[104173.489867] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4/4-0:1.0 (usb)
KERNEL[104173.489934] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4 (usb)
KERNEL[104173.489998] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0 (pci)
[... cut ...]
UDEV  [104189.014014] remove   /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/pci_bus/0000:39 (pci_bus)
UDEV  [104189.014589] remove   /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0 (pci)
UDEV  [104189.014670] remove   /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0 (pci)
UDEV  [104189.014751] remove   /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0 (pci)
UDEV  [104189.014802] remove   /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0 (pci)
UDEV  [104189.015274] remove   /devices/pci0000:00/0000:00:1c.4/0000:03:00.0 (pci)
```

rickysarraf, 2020-02-26:
And this one, an add event, for a USB mouse
```

KERNEL[104799.554493] change   /devices/platform/USBC000:00/typec/port0 (typec)
KERNEL[104799.554589] add      /devices/platform/USBC000:00/typec/port0/port0-partner (typec)
UDEV  [104799.559079] change   /devices/platform/USBC000:00/typec/port0 (typec)
UDEV  [104799.561896] add      /devices/platform/USBC000:00/typec/port0/port0-partner (typec)
KERNEL[104800.342694] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/wakeup/wakeup45 (wakeup)
KERNEL[104800.342775] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/wakeup/wakeup46 (wakeup)
KERNEL[104800.345217] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0 (pci)
KERNEL[104800.345355] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/pci_bus/0000:04 (pci_bus)
KERNEL[104800.468079] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:26/wakeup/wakeup47 (wakeup)
KERNEL[104800.468326] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0 (pci)
KERNEL[104800.468814] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:27/wakeup/wakeup48 (wakeup)
KERNEL[104800.469083] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0 (pci)
UDEV  [104800.469153] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/wakeup/wakeup45 (wakeup)
KERNEL[104800.474378] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/wakeup/wakeup49 (wakeup)
UDEV  [104800.474429] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/wakeup/wakeup46 (wakeup)
KERNEL[104800.474474] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0 (pci)
KERNEL[104800.474529] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0 (pci)
KERNEL[104800.474562] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/pci_bus/0000:05 (pci_bus)
KERNEL[104800.474593] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0/pci_bus/0000:06 (pci_bus)
KERNEL[104800.474622] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/pci_bus/0000:39 (pci_bus)
KERNEL[104800.474652] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/device:29/wakeup/wakeup50 (wakeup)
KERNEL[104800.474706] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0 (pci)
KERNEL[104800.474740] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0/pci_bus/0000:3a (pci_bus)
KERNEL[104800.474791] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0 (pci)
KERNEL[104800.474823] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:04:00.0:pcie210 (pci_express)
KERNEL[104800.474872] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0 (pci)
UDEV  [104800.474920] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:26/wakeup/wakeup47 (wakeup)
KERNEL[104800.474978] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0/0000:04:01.0:pcie210 (pci_express)
KERNEL[104800.475048] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:01.0 (pci)
KERNEL[104800.475108] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:04:02.0:pcie210 (pci_express)
KERNEL[104800.475174] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0 (pci)
KERNEL[104800.475220] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0/0000:04:04.0:pcie210 (pci_express)
KERNEL[104800.475285] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:04.0 (pci)
KERNEL[104800.477518] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3 (usb)
KERNEL[104800.477594] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-0:1.0 (usb)
UDEV  [104800.478161] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:27/wakeup/wakeup48 (wakeup)
KERNEL[104800.478228] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-0:1.0 (usb)
KERNEL[104800.478290] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3 (usb)
KERNEL[104800.478335] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/wakeup/wakeup51 (wakeup)
KERNEL[104800.482494] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4 (usb)
UDEV  [104800.482623] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0 (pci)
KERNEL[104800.482701] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4/4-0:1.0 (usb)
UDEV  [104800.482765] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/wakeup/wakeup49 (wakeup)
KERNEL[104800.482826] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4/4-0:1.0 (usb)
KERNEL[104800.482892] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb4 (usb)
KERNEL[104800.482961] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0 (pci)
UDEV  [104800.483566] add      /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/device:25/device:28/device:29/wakeup/wakeup50 (wakeup)
UDEV  [104800.485610] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/pci_bus/0000:04 (pci_bus)
UDEV  [104800.486287] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0 (pci)
[... cut ...]
UDEV  [104803.334141] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-1/3-1:1.0/0003:093A:2510.0007/input/input52/event20 (input)
UDEV  [104803.337256] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-1/3-1:1.0/0003:093A:2510.0007 (hid)
UDEV  [104803.340436] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-1/3-1:1.0 (usb)
UDEV  [104803.349932] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-1 (usb)
```

rickysarraf, 2020-02-26:
I don't see `laptop-mode-tools` causing problems in device discovery and processing. In initial integration with systemd, we did have some issues on lmt's runtime, which if high could lead udev to kill the process. But that's a completely different behavior in between `udev <=> lmt` and has long been fixed.

By the time, lmt is processing your device (which where it just tries to switch on power saving knobs, if available), your device is supposed to be available and ready.

I can't say for sure if the device's settling is hindered by lmt. It may just be giving you an impression given its runtime cost but I can't see any reason how it'd hinder

rickysarraf, 2020-02-26:
>What I suspect is that the add event for the USB device itself delays processing of the USB interfaces/subdevices, such as the serial device. Here's what usbadm monitor gives when plugging in with the default rules:

Yes. THat could be a possibility. Depending on how many USB type sub-devices are generated. And how many `add` events they generate, this could cause many invocations.

In lmt, to ensure that we don't lose events, while at the same time to ensure that we don't get DoS-ed with too many events, we have a 2 lock system.

The first event processing acquires lock-1, any follow-up event gets to acquire lock-2. All other events during that time span are let go off. When first event (that acquired lock-1) is complete, the second event (that acquired lock-2) is run. The average runtime that I calculated for an lmt invocation was around 4 seconds. But that was very limited to my setup.

And the event that acquired lock-2 will loop for 10 seconds to see that event one is complete and that is can now acquire lock-1.

This is by design, to ensure we don't lose events to process.

matthijskooijman, 2020-02-26:
> I did a plug/unplug of my USB Type-C headphone and I don't get similar results.

It seems you don't have this cgroup stuff, but that might just be related to *what* lmt does exactly on my system. I do not think the cgroup stuff actually causes the problems, it's just an indication that lmt is doing *something*, the real problem is the delay it causes.

In your last post about the mouse, I see a similar delay (over 2s) between the first time the 3-1 usb device is mentioned (the kernel add event) and the last time it is mentioned (the udev add complete event I think?):

``` 
KERNEL[104800.962586] add      /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-1 (usb)
...
UDEV  [104803.349932] bind     /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-1 (usb)
```

> By the time, lmt is processing your device (which where it just tries to switch on power saving knobs, if available), your device is supposed to be available and ready.

It looks like the kernel has made the device available, but udev has not finished processing, so the permissions on the /dev/ file are not set yet. I would think that something similar might happen to e.g. a USB mouse as well 

> Yes. THat could be a possibility. Depending on how many USB type sub-devices are generated. And how many add events they generate, this could cause many invocations.

Ah, I had not actually considered multiple invocations yet, I guess that could indeed be relevant as well. Doing a quick test indeed shows that it runs twice for my USB mouse, and four times for my serial device (probably because it has a few subdevices). If I modify the script to run only the first time, the delay I see in udevadm is about 1s, when it runs twice for the mouse its just under 2s, and for my serial device it was about 3s, so I suspect it's just running multiple times and the delay just stacks). I was thinking the locking might be influential (that the first invocation prevents the second invocation from returning quickly or something), but I don't think that is the case.

I did a quick test, where I let `lmt-udev` run only once for the first device, but added a big delay. E.g. at the top of `lmt-udev`, I added:

```
if [ -e /tmp/lmt ]; then
        exit 0;
fi

echo $$ >> /tmp/lmt
sleep 10
```

When then plugging in an USB mouse (taking care that `/tmp/lmt` does not exist), it takes 10 seconds for the mouse to become usable, so I *think* this shows that any longer runtime that lmt needs is problematic?

rickysarraf, 2020-02-27:
> 
> When then plugging in an USB mouse (taking care that `/tmp/lmt` does not exist), it takes 10 seconds for the mouse to become usable, so I _think_ this shows that any longer runtime that lmt needs is problematic?

Possibly, yes. 
Now, after all this run through,  I am not sure of the processing order of the rules by udev. 

But yes, from what you are showing, if that is the case (runtime length of lmt), then this is an ugly trade-off.

rickysarraf, 2020-02-27:
> Ah, I had not actually considered multiple invocations yet, I guess that could indeed be relevant as well. Doing a quick test indeed shows that it runs twice for my USB mouse, and four times for my [...]

In a short span of time, example a plug/unplug of a device, it usually should settle with 2 invocations. All other event invocations will be ignored. The reason I explained in one of the previous comments.

If you have more than 2 instances of the run, I'd be interested in the time span. Because the kernel can generate many many events on device plugs/unplugs, but because we do a global power savings settings re-apply, we only try to honor 2 instances (**in the short time span**)

`lmt-udev force` will re-apply settings on all devices. Not any specific one. The newly plugged device gets covered in it.

PS: I hope I have been able to explain in words. /me realizes my English still needs a lot of improvement :-)

matthijskooijman, 2020-02-27:
>  Now, after all this run through, I am not sure of the processing order of the rules by udev.

Yeah, this might be something to further investigate.

> If you have more than 2 instances of the run, I'd be interested in the time span.

I just counted the invocations of `lmt-udev`, I did not check if anything actually *happens* based on that invocation. In fact, your assertion that only 2 runs execute would make sense to explain that the delay is on 3s (not 4s) with 4 invocations.

> PS: I hope I have been able to explain in words. /me realizes my English still needs a lot of improvement :-)

I've had no problem following your English, so don't worry :-)