rickysarraf/laptop-mode-tools #143: Sometimes gets stuck with WD5000LPCX-2 HDD

paradox-ptr, 2019-02-26:
Here's an annoying one.

After fixing #142, to make sure that plug change works, if I remove power plug during system shutdown (laptop mode service running), shutdown process reliably gets stuck and HDD starts making noises (periodic eigth-eigth-forth on repeat).

Also, once this has happened on boot with power plug disconnected, though boot getting stuck I'm unable to reproduce.

I don't have hdparm installed. But I tried to install it and run hdparm -B 1 /dev/sda, and it gets stuck in exactly the same way, and makes the same sounds.

Regular stop/restart of laptop_mode service works just fine.

Boot and system drive on my system is M.2 SSD, HDD is this one:
```
# lshw -class disk  
Password: 
  *-disk                    
       description: ATA Disk
       product: WDC WD5000LPCX-2
       vendor: Western Digital
       physical id: 0.0.0
       bus info: scsi@2:0.0.0
       logical name: /dev/sda
       version: 1A02
       serial: WD-<cut>
       size: 465GiB (500GB)
       capabilities: gpt-1.00 partitioned partitioned:gpt
       configuration: ansiversion=5 guid=<cut> logicalsectorsize=512 sectorsize=4096
```
laptop-mode-tool is 1.72.2-r1, from Gentoo repo, kernel is 4.20.12. Laptop is Lenovo Ideapad 700-15ISK.

The drive already does head parking, so I'd rather software not mess around with it, and let HW handle this. So I set HD="" in laptop-mode.conf, but issue persists.

rickysarraf, 2019-05-24:
Your issue sounds more like #123 but you are on a recent enough kernel that most fixes should have come in. Without any logs, it is hard to predict what must be going wrong...

If you want to root cause the issue, you may want to try blacklisting the device id into `runtime-pm.conf` one by one.

rickysarraf, 2019-12-09:
No further message from reported. Please re-open bug if you have the relevant logs

closed 2019-12-09 by rickysarraf