rickysarraf/laptop-mode-tools #144: can't start lmt-config-gui

doomdevelop, 2019-02-28:
Surface Pro 2017 Ubuntu 18.10.
`pkexec /usr/sbin/lmt-config-gui
Using PyQt4
No protocol specified
lmt.py: cannot connect to X server :0
`
Probably related to the changes ubuntu made for gui apps with root permission.
Any idea how I can fix the command ?

Thank you !

rickysarraf, 2019-06-03:
I am sorry for the very late reply.

It could possibly be because of `Wayland`. But I am not sure but iirc the last time I had tested on `Wayland` too.

Here, on `Xorg`, it is working fine.

[image]

Higurashi3301, 2019-08-20:
Since Wayland doesn't allow graphical applications to run with sudo, you need to make it fallback to xwayland via `xhost si:localuser:root` and than you can run it normally with `sudo lmt-config-gui` (and to remove the fallback, type `xhost -si:localuser:root`).

FrostKnight, 2019-11-17:
Could also be related to this:  https://github.com/rickysarraf/laptop-mode-tools/issues/124#issue-294213802

I had that issue a while back... ;)

rickysarraf, 2021-06-10:
On today's Wayland implementation on KDE, this is working out of the box. The big major difference than what was initially reported is that this is now based on PyQT5. Not sure if that has a role to play or not, but this issue isn't valid on my Wayland setup.

[image]

closed 2021-06-10 by rickysarraf