rickysarraf/laptop-mode-tools #145: battery-level-polling causes slow boot

xaltsc, 2019-03-16:
OS: VoidLinux with kernel 4.19.29_1 and LVM on LUKS

When starting up with laptop-mode enabled and battery-level-polling, starting up takes about two more minutes than usual. With loglevel=5 as boot option, udev shows: `udevd: slow 'lmt-udev force'`.

I haven't been able to find other helpful messages sadly.

rickysarraf, 2019-03-18:
I have the same identical setup here (laptop-mode and battery-level-polling enabled) running on Debian without any issues.

I need more information to debug the issue, if any.  Can you please set: `VERBOSE_OUTPUT=1` in the `/etc/laptop-mode/laptop-mode.conf` configuration file and then reboot ? Hopefully, that'll give much more details to the journal/syslog for us to figure out what is wrong with your setup.

Once you boot, please share the relevant log file.

rickysarraf, 2019-12-09:
No response from reported. Please re-open with necessary information.

closed 2019-12-09 by rickysarraf