rickysarraf/laptop-mode-tools #146: cpufreq medium value

GFdevelop, 2019-03-27:
I don't have `/sys/devices/system/cpu/cpu*/cpufreq/scaling_available_frequencies`, how to set a medium/custom value? I can set only slowest-fastest or slowest-slowest or fastest-fastest for all governors.
```
~ $ ls -a1 /sys/devices/system/cpu/cpu0/cpufreq/
affected_cpus
cpuinfo_max_freq
cpuinfo_min_freq
cpuinfo_transition_latency
related_cpus
scaling_available_governors
scaling_cur_freq
scaling_driver
scaling_governor
scaling_max_freq
scaling_min_freq
scaling_setspeed
```
```
~ $ cpupower frequency-info
analyzing CPU 0:
  driver: intel_pstate
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency:  Cannot determine or is not supported.
  hardware limits: 800 MHz - 3.60 GHz
  available cpufreq governors: performance powersave
  current policy: frequency should be within 800 MHz and 3.60 GHz.
                  The governor "powersave" may decide which speed to use
                  within this range.
  current CPU frequency: Unable to call hardware
  current CPU frequency: 2.25 GHz (asserted by call to kernel)
  boost state support:
    Supported: yes
    Active: yes
```

```
~ $ cat /etc/laptop-mode/conf.d/cpufreq.conf
...
CONTROL_CPU_FREQUENCY="auto"
...
BATT_CPU_MAXFREQ=fastest
BATT_CPU_MINFREQ=slowest
BATT_CPU_GOVERNOR=powersave
BATT_CPU_IGNORE_NICE_LOAD=1
LM_AC_CPU_MAXFREQ=fastest
LM_AC_CPU_MINFREQ=slowest
LM_AC_CPU_GOVERNOR=performance
LM_AC_CPU_IGNORE_NICE_LOAD=0
NOLM_AC_CPU_MAXFREQ=fastest
NOLM_AC_CPU_MINFREQ=slowest
NOLM_AC_CPU_GOVERNOR=performance
NOLM_AC_CPU_IGNORE_NICE_LOAD=0
...
```

Thanatermesis, 2019-10-03:
I have a patch for this :) 

[patch.diff.gz](https://github.com/rickysarraf/laptop-mode-tools/files/3688462/patch.diff.gz)

rickysarraf, 2021-06-10:
Your's is a special CPU then. I will adapt @Thanatermesis patch

closed 2021-06-10 by rickysarraf