rickysarraf/laptop-mode-tools #147: System hang after resume on battery with LMT enabled.

PF4Public, 2019-04-05:
Recently (after installing LMT) I've noticed that my laptop hangs upon resume when on battery. Through trial and error I found that it takes one switch from AC to battery to make this happen. After that even stopping LMT daemon does not help.

Turns out, LMT do set `/sys/bus/scsi/devices/0:0:0:0/power/autosuspend_delay_ms` unconditionally to `0`. By itself it is not a trouble at all, if `control` is `on`, but should `control` become `auto`, it for some reason hangs the whole system upon resume from suspend. 

By default `autosuspend_delay_ms` is set to `-1`. With this setting suspending and resuming works fine with AC or battery. But LMT for some reason overwrites `autosuspend_delay_ms` with `0` even if the `sd` is in blacklist. After that, when switching to battery, LMT writes `auto` in `control`. With `auto` in `control` and `0` in `autosuspend_delay_ms` system hangs upon resume from suspend. This does not happen when `control` is `on` and `autosuspend_delay_ms` is `0`.

My system is running `4.19.27-gentoo-r1` kernel.

I have added another check for `if ! blacklisted $runtime_device; then USE_DEVICE=1` in order for LMT to skip changing  `autosuspend_delay_ms` for blacklisted `sd` device. This way  `autosuspend_delay_ms` is kept at its default value: `-1` and the hang upon resume from suspend is no more. Is there any better solution? And more importantly, why is it happening?

It could be somehow related to #123

rickysarraf, 2019-04-06:
As mentioned before, please enable these settings and then see if it generates some useful logs. We need to look at what is happening underneath. Please set `VERBOSE_OUTPUT` to `1`

```
# Set this to 1 if you want to see a lot of information when you start/stop 
# laptop_mode.
#
VERBOSE_OUTPUT=0

# Set this to 1 if you want to log messages to syslog
LOG_TO_SYSLOG=1
```

PF4Public, 2019-04-13:
Sorry for a long delay

I've commented out my additions to runtime-pm script

With this script reverted turning off AC gives following logs: https://pastebin.com/uFvDTFZq
And giving back power gives: https://pastebin.com/YxK28KyU

rickysarraf, 2019-04-15:
@PF4Public These do not include system (kernel) logs. Can you please share the full logs ?

PF4Public, 2019-04-15:
Not much happened actually:

https://pastebin.com/ARHs7Wau

https://pastebin.com/eLKrwkhC

rickysarraf, 2019-05-24:
Is this all that was logged for the kernel ?
There's nothing much in here:

```
Apr 13 23:17:21 laptop kernel: [28543.871655] usb 1-1.2: new full-speed USB device number 24 using ehci-pci
Apr 13 23:17:21 laptop kernel: klogd 1.5.1, ---------- state change ---------- 
Apr 13 23:17:21 laptop kernel: No module symbols loaded - kernel modules not enabled. 
Apr 13 23:17:21 laptop kernel: [28543.981532] usb 1-1.2: New USB device found, idVendor=09da, idProduct=9090, bcdDevice= 1.02
```
The only thing I can notice in here is that it seems you have got a monolithic kernel with no loadable modules support. Is it so ? If yes, why so ?

PF4Public, 2019-05-24:
>Is this all that was logged for the kernel ?

Yep

>The only thing I can notice in here is that it seems you have got a monolithic kernel with no loadable modules support. Is it so ? If yes, why so ?

No, that's untrue. Just a misleading message from kernel. Loadable modules are supported and used.

>There's nothing much in here:

That is exactly, what I was trying to convey.
Nothing bad is happening while LMT is _changing modes_ (AC-Battery).
The _only_ bad thing is that 
>Turns out, LMT do set /sys/bus/scsi/devices/0:0:0:0/power/autosuspend_delay_ms unconditionally to 0. By itself it is not a trouble at all, if control is on, but should control become auto, it for some reason hangs the whole system upon resume from suspend.

jhyeon, 2019-07-17:
I have also been suffering from the hang after resume on ac following suspend on battery.
@PF4Public thank you very much for finding out the reason!  Indeed, the problem disappears if LMT is amended not to touch at all autosuspend parameters of blacklisted devices.  IMO, it is what blacklists are for.

closed 2019-07-24 by rickysarraf