rickysarraf/laptop-mode-tools #148: LCD Brightness module causes a loop of harddrive syncs

jameh, 2019-05-07:
System log after unplugging my power cable: https://hastebin.com/uperedilev

Brightness of my lcd is lowered immediately.

I can hear harddrive activity every second in sync with those SCSI messages.

it stops after a bit, but it seems that if I replug the power before it is done vomitting those messages, the brightness is not updated right away - like it's causing laptop_mode to become unresponsive

I'm running Arch Linux. Here's my module config file for the LCD control:
https://hastebin.com/aqabilonir

rickysarraf, 2019-05-24:
Hi... Your logs do not seem to be accessible. Can you please attach them directly to this task instead ?

rickysarraf, 2019-12-09:
No response from reported. When you have the necessary information, please attach it to this bug report on Github. Remote location links are useless for the very reason, which is now, where I can't access them.

closed 2019-12-09 by rickysarraf