rickysarraf/laptop-mode-tools #153: Bluetooth conf not being recognized

reikdas, 2019-11-20:
This is my bluetooth.conf - 
```
#
# Configuration file for Laptop Mode Tools module bluetooth.
#
# For more information, consult the laptop-mode.conf(8) manual page.
#

###############################################################################
# Bluetooth settings
# ------------------
#
#__COMMENT If you enable this module, laptop mode tools will enable/disable bluetooth
#__COMMENT depending on the power status of your laptop. Bluetooth uses a considerable
#__COMMENT amount of power (comparable to wireless networking), and disabling it is
#__COMMENT therefore a good idea when you are looking to improve your battery life.
#
###############################################################################

# Enable debug mode for this module
# Set to 1 if you want to debug this module
DEBUG=0

# Control bluetooth?
CONTROL_BLUETOOTH=0

# Enable bluetooth on battery
BATT_ENABLE_BLUETOOTH=1

# Enable bluetooth on AC
AC_ENABLE_BLUETOOTH=1

# Bluetooth interfaces to enable/disable
BLUETOOTH_INTERFACES="hci0"
```
As is evident, bluetooth control is disabled. 

But when my laptop is using laptop-mode-tools and is below a certain threshold (I think 50%), it keeps disconnecting my bluetooth devices. 
I checked that this is indeed because of laptop-mode-tools by disabling it and everything worked perfectly.

rickysarraf, 2019-12-09:
It would depend on multiple factors. Many hardware these days, embed the bluetooth and wifi onto a single chip. In such cases, applying power savings to your wifi could affect your bt device too. But all depends on the hardware.

Can you please investigate and provide:
* system logs when bt is behaving erratic
* Laptop Mode Tools logs, preferably in verbose mode.
* Laptop Mode Tools version, Distribution, Linux Kernel version

/me needs to read up on bug templates for Github.

rickysarraf, 2021-06-10:
Please re-open when you have the necessary information. :pray:

closed 2021-06-10 by rickysarraf