rickysarraf/laptop-mode-tools #154: wireless mouse causing error in laptop_mode file

Bulmer, 2019-12-08:
I had a "/sys/class/power_supply/hidpp_battery_0/present - file not found" error which occurred because I have a wireless usb mouse connected which apparently reports the mouse battery in /sys/class/power_supply/ as "hidpp_battery_0" but does not report it as present because its not locally present. I have temporarily fixed it by placing another check in checkBattery() [L669].
      elif [ ! -f $BATT/present ]; then
        log "VERBOSE" "HID Mouse Battery, skipping."

Could you check that out and maybe add it to the source?

rickysarraf, 2019-12-09:
What version of `laptop-mode-tools` are you on ? We are already doing a check for that file to determine if we should act on it ?

https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/usr/sbin/laptop_mode#L759
`				PRESENT=$(cat $BATT/present 2> /dev/null || echo "0")`

Bulmer, 2019-12-09:
Version 1.72-3 and am running it on Debian Testing.

rickysarraf, 2019-12-11:
So this is reported and fixed with: 0311b5d69d4b273a53f6f18213a62fba69862f74
I will do a release soon. Meanwhile, you can patch it yourself and the change is just a one-liner.

closed 2019-12-11 by rickysarraf

BillyCroan, 2024-01-20:
I'm having the same problem, but for me present=1
I don't think it's the intention of laptop-mode-tools to react to peripheral batteries.  So perhaps there's a better way to screen them out.

Why for instnace, are we enumerating /sys/class/power_supply/* when we could enumerate /sys/class/power_supply/BAT* ?

BillyCroan, 2024-01-20:
I should also note, that this stylus pen, always reports that it is discharging, 1% remaining.  That doesn't help anything, I'm sure.  Can LMT simply ignore all peripheral batteries?

BillyCroan, 2024-01-20:
I should also note, that this stylus pen, after the first time i used it after a fresh boot, always reports that it is discharging, 1% remaining.  That doesn't help anything, I'm sure.  Can LMT simply ignore all peripheral batteries?

Also when you print to syslog these warnings:
2024-01-20T12:08:38.150322-06:00 x360 laptop_mode[187670]: WARNING: Battery does not report a capacity. Minimum battery
2024-01-20T12:08:38.150388-06:00 x360 laptop_mode[187670]: charge checking does not work without a design capacity.
2024-01-20T12:08:38.150406-06:00 x360 laptop_mode[187670]: WARNING: Battery does not report a design capacity. Auto hibernation
 
can you include the NAME of the offending battery to narrow down resolution?