rickysarraf/laptop-mode-tools #156: Can laptop mode write case only inside certain directories or mount options?

brunoais, 2020-05-29:
I tried to confirm that information through the configuration file but the file only shows device whitelist and blacklist.
Is there a mount option (or a set of them) so I can force certain data to delay writes?
For example, I want to force `/tmp` to have delayed writes instead of tmpfs but I don't want a document I just successfully saved in `$HOME/Documents` to be lost.

I'm fine with using `mount` options or `chattr` for this purpose.

rickysarraf, 2020-05-30:
fiddling with tmpfs writeback cache timers is pointless. tmpfs in ram based. so delaying logic doesn't apply there.

For any special cases, you can always use the module `exec-commands`. Please refer to man page for details.

closed 2020-05-30 by rickysarraf

brunoais, 2020-05-30:
This is not about fiddling with tmpfs.
I even mention: 

> delayed writes **instead of** tmpfs 

I want to use a different file system than `tmpfs` but, for that, I want to have delayed writes instead. Maybe I can use ext4 or xfs for the `/tmp` but I want data that stays there for long enough to be eventually written down.
Did I clarify it right @rickysarraf ?

rickysarraf, 2020-06-01:
Okay. I think I get it now. You are looking for a way to define `per-mount-point` timeout values ?
Right now, lmt does it globally for all mount points.

But I think these are the exact kind of things, for which we have the `exec-commands` module.

brunoais, 2020-06-01:
How can I make use of `exec-commands` to create different timeouts for each mount/directory/process?
From my understanding, `exec-commands` is only about executing commands when a power state changes between AC and battery. That's also what I read in the manual:

> trigger the command during power state changes

Is there some command, flag or virtual file I'm missing that allows this kind of change that I don't know of which lmt changes to alter the writes?

rickysarraf, 2020-06-02:
`exec-commands` module is flexible enough to serve the special needs of users.

Here's an example snippet.
```
BATT_EXEC_COMMAND_0="" # Enable power saving feature 0 here
LM_AC_EXEC_COMMAND_0="" # Enable power saving feature 0 here
NOLM_AC_EXEC_COMMAND_0="" # Disable power saving feature 0 here
```

So, say, you want to set timeout value for `/home` to be 900 seconds, when on battery. Then your configuration will be like:

```
BATT_EXEC_COMMAND_0="mount -o remount,commit=900 /home"
LM_AC_EXEC_COMMAND_0="mount -o remount,commit=0 /home"
NOLM_AC_EXEC_COMMAND_0=""
```