rickysarraf/laptop-mode-tools #161: Insert USB device causes eth0 be disabled

akikaede6, 2021-01-13:
**Describe the bug**
My laptop has two network port, named eth0 and eth1, and my battery driver has some problems, it doesn't have ac info, laptop-mode always identify it using battery, when i insert USB device, my eth0 network will be down.

**Here is my /etc/laptop-mode/conf.d/ethernet.conf**
```
# Note: If you don't know how to find the speed of your ethernet card, leave the                                        
# following values as it is.                                                                                            
THROTTLE_SPEED="slowest"                                                                                                                                                                                                                        # Disable wakeup-on-LAN? This will disable wakeup-on-LAN unconditionally,                                               
# independent of battery power. It will save power while your laptop is turned                                          
# off or in standby mode.                                                                                               
DISABLE_WAKEUP_ON_LAN=1                                                                                                                                                                                                                         # A list of the ethernet devices that should be controlled.                                                             
ETHERNET_DEVICES="eth0"                                                                                                                                                                                                                                                                                                                                                 # In practise, most of the times a user is on battery, she is using the wireless device                                 
# Under such cases, you might want to disable your ethernet interface completely, when                                  
# on battery.
#                                                                                                           
# Set below setting to 1, to disable ethernet interface when on battery and                                            
# when no carrier is detected on the interface (e.g., no active cable is                                                
# plugged in).                                                                                                          
DISABLE_ETHERNET_ON_BATTERY=1
```

I test my another laptop, when I unlplug the battery adapter, my eth0 be down, it was expected, and I use `ifconfig eth0 up`, then I insert the USB device, it makes my eth0 down too.
I fond that in `/lib/udev/rules.d/99-laptop-mode.rules`, `ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force"`, then lmt-udev force will exec `/usr/sbin/laptop_mode`, load `/etc/laptop-mode/conf.d`, it makes my eth0 down, but it confuses me, why insert USB device will get my eth0 down? does it save power?

**Important Information:**
 - Linux Kernel
 one 4.19.0
another 4.4.15
 - Laptop Mode Tools version
one 1.72.3
another 1.72.2

rickysarraf, 2021-01-13:
It could be for various reasons depending on what hardware you have. Is your Ethernet LOM or is it on the USB bus ? What driver does is use ? What does dmesg/journalctl say about the error message ?

There are just too many questions. Once you answer the above ones, we may be in a better position to make a guess.

akikaede6, 2021-01-14:
Thank you for reply! My test computer is macbook pro 2011 with ubuntu 18.04, one hour before I test it in my friend's thinkpad T490 whit ubuntu 20.04, USB event doesn't affect network, maybe ubuntu have fixed it, I'll try ubuntu 20.04 in my macbook pro later.

akikaede6, 2021-01-14:
I tried ubuntu 20.04 in my macbook pro, it's network got down too.Here's journalctl -f massage following, added `loglevel=7 systemd.log_level=debug debug=y` in grub, network got down in 14:24:49
```
1月 14 14:24:46 aki-MacBookPro kernel: usb 2-1.2: new high-speed USB device number 5 using ehci-pci
1月 14 14:24:46 aki-MacBookPro kernel: usb 2-1.2: New USB device found, idVendor=14cd, idProduct=6116, bcdDevice= 2.20
1月 14 14:24:46 aki-MacBookPro kernel: usb 2-1.2: New USB device strings: Mfr=1, Product=3, SerialNumber=2
1月 14 14:24:46 aki-MacBookPro kernel: usb 2-1.2: Product: USB Mass Storage Device
1月 14 14:24:46 aki-MacBookPro kernel: usb 2-1.2: Manufacturer: Generic
1月 14 14:24:46 aki-MacBookPro kernel: usb 2-1.2: SerialNumber: 116AC2101219
1月 14 14:24:46 aki-MacBookPro kernel: usb-storage 2-1.2:1.0: USB Mass Storage device detected
1月 14 14:24:46 aki-MacBookPro kernel: scsi host6: usb-storage 2-1.2:1.0
1月 14 14:24:46 aki-MacBookPro systemd-udevd[12524]: value '[dmi/id]sys_vendor' is 'Apple Inc.'
1月 14 14:24:46 aki-MacBookPro systemd-udevd[12524]: value '[dmi/id]sys_vendor' is 'Apple Inc.'
1月 14 14:24:46 aki-MacBookPro systemd-udevd[12524]: Successfully forked off '(spawn)' as PID 12525.
1月 14 14:24:46 aki-MacBookPro mtp-probe[12525]: checking bus 2, device 5: "/sys/devices/pci0000:00/0000:00:1d.7/usb2/2-1/2-1.2"
1月 14 14:24:46 aki-MacBookPro mtp-probe[12525]: bus: 2, device: 5 was not an MTP device
1月 14 14:24:46 aki-MacBookPro systemd-udevd[12524]: created db file '/run/udev/data/c189:132' for '/devices/pci0000:00/0000:00:1d.7/
usb2/2-1/2-1.2'
1月 14 14:24:46 aki-MacBookPro systemd-udevd[12524]: Successfully forked off '(spawn)' as PID 12526.
1月 14 14:24:46 aki-MacBookPro laptop-mode[12580]: enabled, active
1月 14 14:24:46 aki-MacBookPro systemd[1]: Got message type=method_call sender=:1.104 destination=org.freedesktop.systemd1 path=/org/
freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=LookupDynamicUserByUID cookie=2 reply_cookie=0 signature=u err
or-name=n/a error-message=n/a
1月 14 14:24:46 aki-MacBookPro systemd[1]: Sent message type=method_return sender=n/a destination=:1.104 path=n/a interface=n/a membe
r=n/a cookie=746 reply_cookie=2 signature=s error-name=n/a error-message=n/a
1月 14 14:24:46 aki-MacBookPro systemd[1]: Got message type=method_call sender=:1.105 destination=org.freedesktop.systemd1 path=/org/
freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=LookupDynamicUserByUID cookie=2 reply_cookie=0 signature=u err
or-name=n/a error-message=n/a
1月 14 14:24:46 aki-MacBookPro systemd[1]: Sent message type=method_return sender=n/a destination=:1.105 path=n/a interface=n/a membe
r=n/a cookie=747 reply_cookie=2 signature=s error-name=n/a error-message=n/a
1月 14 14:24:46 aki-MacBookPro kernel: tg3 0000:02:00.0 enp2s0f0: Link is down
1月 14 14:24:46 aki-MacBookPro systemd-resolved[9233]: Removing scope on link enp2s0f0, protocol dns, family *
1月 14 14:24:46 aki-MacBookPro systemd[1]: systemd-resolved.service: Got notification message from PID 9233 (WATCHDOG=1)
1月 14 14:24:46 aki-MacBookPro kernel: EXT4-fs (sda2): re-mounted. Opts: errors=remount-ro,commit=600
1月 14 14:24:47 aki-MacBookPro kernel: scsi 6:0:0:0: Direct-Access        Mass  Storage Device        PQ: 0 ANSI: 0
1月 14 14:24:47 aki-MacBookPro kernel: sd 6:0:0:0: Attached scsi generic sg2 type 0
1月 14 14:24:47 aki-MacBookPro kernel: sd 6:0:0:0: [sdb] 937703086 512-byte logical blocks: (480 GB/447 GiB)
1月 14 14:24:47 aki-MacBookPro kernel: sd 6:0:0:0: [sdb] Write Protect is off
1月 14 14:24:47 aki-MacBookPro kernel: sd 6:0:0:0: [sdb] Mode Sense: 03 00 00 00
1月 14 14:24:47 aki-MacBookPro kernel: sd 6:0:0:0: [sdb] No Caching mode page found
1月 14 14:24:47 aki-MacBookPro kernel: sd 6:0:0:0: [sdb] Assuming drive cache: write through
1月 14 14:24:47 aki-MacBookPro systemd-udevd[13610]: value '[dmi/id]sys_vendor' is 'Apple Inc.'
1月 14 14:24:47 aki-MacBookPro systemd-udevd[13610]: value '[dmi/id]sys_vendor' is 'Apple Inc.'
1月 14 14:24:47 aki-MacBookPro systemd-udevd[13610]: passed 138 byte device to netlink monitor 0x555af4fa4cd0
1月 14 14:24:47 aki-MacBookPro kernel:  sdb: sdb1 sdb2 sdb3
1月 14 14:24:47 aki-MacBookPro kernel: sd 6:0:0:0: [sdb] Attached SCSI disk
1月 14 14:24:48 aki-MacBookPro root[14307]: Device /sys/bus/usb/devices/1-0:1.0 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14326]: Device /sys/bus/usb/devices/1-1 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14343]: Device /sys/bus/usb/devices/1-1.1 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14371]: Device /sys/bus/usb/devices/1-1.1:1.0 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14378]: Device /sys/bus/usb/devices/1-1.2 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14382]: Device /sys/bus/usb/devices/1-1.2:1.0 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14386]: Device /sys/bus/usb/devices/1-1.2:1.1 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14393]: Device /sys/bus/usb/devices/1-1:1.0 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14416]: Device /sys/bus/usb/devices/2-0:1.0 is blacklisted, skipping auto suspend.
1月 14 14:24:48 aki-MacBookPro root[14425]: Device /sys/bus/usb/devices/2-1 is blacklisted, skipping auto suspend.
[... cut ...]
1月 14 14:24:50 aki-MacBookPro systemd[1]: dev-disk-by\x2dpartuuid-ef5376dc\x2d023f\x2d44fe\x2d8739\x2d9ba3baed2a7d.device: Changed $
```
Here is my network hardware, I'm sorry that I don't know how to check if my ethernet LOM or in the USB bus, I found that my macbook pro wired net is connected by RJ-45.
```
02:00.0 Ethernet controller: Broadcom Inc. and subsidiaries NetXtreme BCM57765 Gigabit Ethernet PCIe (rev 10)
03:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4331 802.11a/b/g/n (rev 02)
```

akikaede6, 2021-01-15:
Does this conf `ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force"` in `/lib/udev/rules.d/99-laptop-mode.rules` lack of consideration？USB event will force to load laptop-mode conf, it will set all the config that set on, not only network conf, I fond that my DPMS also set to 300 when I insert USB device, maybe it will cause other problems.

rickysarraf, 2021-01-15:
> Thank you for reply! My test computer is macbook pro 2011 with ubuntu 18.04, one hour before I test it in my friend's thinkpad T490 whit ubuntu 20.04, USB event doesn't affect network, maybe ubuntu have fixed it, I'll try ubuntu 20.04 in my macbook pro later.

Thanks. Every machine is different. And depending on what bus the device is, and how good/crappy the device driver is, there will be some side-effects. As a data point, you've already known now that this issue is not see on another machine.

rickysarraf, 2021-01-15:
> Does this conf `ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force"` in `/lib/udev/rules.d/99-laptop-mode.rules` lack of consideration？USB event will force to load laptop-mode conf, it will set [...]

It shouldn't cause any problems as such. There are only 2 states that LMT cares about, `ON_BATT` and `ON_AC`. When `ON_BATT`, it does a bunch of power saving tweaks. The additional rule is just to re-enforce the same, when you plug-in any new usb device.

rickysarraf, 2021-01-15:
The behavior of losing the network, has probably to do with your hardware. As in, when you switch to power saving mode, the network chip resets. I recollect having such reports in the past, which is why it led me to disable ethernet power saving, by default.

Yes. I can confirm that in https://github.com/rickysarraf/laptop-mode-tools/blob/bc4bf873fab2966f115836d6c410f994b9e398cd/etc/laptop-mode/conf.d/ethernet.conf#L26

Can you set `CONTROL_ETHERNET` to `0` in `/etc/laptop-mode/conf.d/ethernet.conf` and then reboot and do the same usb plug/unplug exercise ? I think that'll solve your problem.

akikaede6, 2021-01-15:
> The behavior of losing the network, has probably to do with your hardware. As in, when you switch to power saving mode, the network chip resets. I recollect having such reports in the past, which is [...]

That works! But I'm wondering if I want my laptop get my eth network down when in a emergy power, and don't want USB event affect it, should I set this conf to auto? In what situation auto mode will start to work?

rickysarraf, 2021-01-15:
> That works! But I'm wondering if I want my laptop get my eth network down when in a emergy power, and don't want USB event affect it, should I set this conf to auto? In what situation auto mode will start to work?

No. `auto` is synonymous to `1`. See this section about `auto modules`
https://github.com/rickysarraf/laptop-mode-tools/blob/bc4bf873fab2966f115836d6c410f994b9e398cd/etc/laptop-mode/laptop-mode.conf#L135

In your case, the hardware is not able to switch to power saving mode without a device reset, which results in losing the network and then reconnecting momentarily. Unfortunately, there's not much `laptop-mode-tools` can do here. That's an oddity of the device driver for your network card.

rickysarraf, 2021-01-20:
@akikaede6 I'm closing this bug report as there's not much laptop-mode-tools can do other than blacklist the device/module. If you have any new findings, please do feel free to re-open this bug.

closed 2021-01-20 by rickysarraf