rickysarraf/laptop-mode-tools #167: Documentation for LM_SECONDS_BEFORE_SYNC is unclear

d-g, 2021-05-16:
Dear maintainers,

regarding `LM_SECONDS_BEFORE_SYNC` laptop-mode.conf(8) says:

> The number of seconds that laptop mode waits after the disk goes idle before it starts a full sync. This should always be less than your hard disk idle timeout, because otherwise you'll have a sync directly after your drive spins down. Two seconds is usually a good value for this option.

Unfortunately, it does not to make much sense when read as written.  If that would indeed be a timeout started after going idle, then in order to avoid syncs right after spindowns, it should be set to a value _more_ that disk idle timeout (i. e. a timeout that as well starts when disk goes idle and spins it down on expiration).

If I understand Linux documentation [1] for `/proc/sys/vm/laptop_mode` (which is what this variable really controls) correctly, itʼs a number of seconds that a machine waits after disk goes _active_, not idle.

[1] file:///usr/share/doc/linux-doc/Documentation/laptops/laptop-mode.txt.gz

rickysarraf, 2021-05-17:
This looks good to me. The only thing, over time, is that these days we commonly have laptops with non-rotational media. But it still applies to an extent. Just not absolute relevance in this era.

A disk activity can happen for both, reads and writes. On spinning disks, an either activity means spinning up the disk. By this knob, we tell the kernel that if can, it should write the dirty blocks whenever the disk has spun up.

This is the kernel documentation.
>Laptop mode is controlled by the knob /proc/sys/vm/laptop_mode. This knob is
present for all kernels that have the laptop mode patch, regardless of any
configuration options. When the knob is set, any physical disk I/O (that might
have caused the hard disk to spin up) causes Linux to flush all dirty blocks. The
result of this is that after a disk has spun down, it will not be spun up
anymore to write dirty blocks, because those blocks had already been written
immediately after the most recent read operation. The value of the laptop_mode
knob determines the time between the occurrence of disk I/O and when the flush
is triggered. A sensible value for the knob is 5 seconds. Setting the knob to
0 disables laptop mode.

rickysarraf, 2021-06-10:
I'm closing this issue. If you think there's more to it, please do feel free to re-open. :pray:

closed 2021-06-10 by rickysarraf