rickysarraf/laptop-mode-tools #168: disable screen blanking support,

FrostKnight, 2021-06-10:
I like most of what laptop-mode-tools does, but how do you stop the screen from blanking every 5 mins?

What do i need to configure to change this problem?

Also, can you add how to do this to the guide for the foreseeable future?

When its idle for even 5 mins the screen turns black till I click some key, and because I have xss-lock on, it can be quite a pain. Aka, xss-lock locks the screen when it dims/blanks. ;)

System, Hyperbola, and btw, I have tried using your latest version, and also Hyperbola's version.

Your documentation on how to do this is unclear.. ;/

rickysarraf, 2021-06-10:
I guess you meant the `linux terminal blanking`.

Based on your feedback in this bug report, I checked to see the documentation on what may be missing. Quoting the manpage:

```
   /etc/laptop-mode/conf.d/terminal-blanking.conf
       The terminal-blanking module allows you to control the terminal blanking timeouts for the Linux text console.

       CONTROL_TERMINAL
                 When this option is enabled, laptop mode will control the terminal blanking settings for Linux's virtual consoles.

       TERMINALS This option should contain a space-separated list of console device files that should be affected by the  terminal  blanking
                 settings.  Only  one console device file needs to be included, because the settings are shared between all virtual consoles.
                 By default this setting is set to /dev/tty1.

       BATT_TERMINAL_BLANK_MINUTES

       LM_AC_TERMINAL_BLANK_MINUTES

       NOLM_AC_TERMINAL_BLANK_MINUTES

       BATT_TERMINAL_POWERDOWN_MINUTES

       LM_AC_TERMINAL_POWERDOWN_MINUTES

       NOLM_AC_TERMINAL_POWERDOWN_MINUTES
                 These options specify the terminal blanking and powerdown timeouts, in minutes. The allowed ranges are 1-60 minutes, or 0 to
                 disable blanking or powerdown. The values are cumulative: the powerdown value is counted from the moment of screen blanking,
                 not from the start of inactivity.
```

Quoting the actual settings from the default config file.

```bash
#
# Should laptop mode tools control terminal blanking settings?
#
# Set to 0 to disable
CONTROL_TERMINAL="auto"

#
# Terminal device files that should be affected. (One terminal is enough, this
# affects all consoles. )
#
TERMINALS="/dev/tty1"

#
# These settings specify the blanking and powerdown timeouts. Note that
# the powerdown timeout is counted from the moment the screen is blanked,
# i.e. BLANK_MINUTES=2 and POWERDOWN_MINUTES=5 means the screen powers
# down after 7 minutes of inactivity. The range for all these settings is
# 1 to 60 minutes, or 0 to disable.
#
BATT_TERMINAL_BLANK_MINUTES=1
BATT_TERMINAL_POWERDOWN_MINUTES=2
LM_AC_TERMINAL_BLANK_MINUTES=10
LM_AC_TERMINAL_POWERDOWN_MINUTES=10
NOLM_AC_TERMINAL_BLANK_MINUTES=10
NOLM_AC_TERMINAL_POWERDOWN_MINUTES=50
```

If you set that value to 0, i.e. `CONTROL_TERMINAL=0`, that should do the job.

Any enhancements to the documentation, is welcome, preferably as a PR to this repository.

FrostKnight, 2021-06-10:
Hmm, thats very odd, I tried this once and it still kept doing it, although maybe its because I removed the apostrophes...

strange, will see if your suggestion works.

FrostKnight, 2021-06-10:
Actually, on further inspection, I think I wasn't clear...

I meant when using a window manager like jwm or i3-wm.

For some reason it keeps suspending every 5 minutes or less. Weird it is... 

I am not even using a terminal at the time.

rickysarraf, 2021-06-11:
Are there good reasons to believe that it is LMT that is the cause of the blanking ? It could be something else altogether. Nevertheless, it will be useful only if you can help generate some more information.

rickysarraf, 2021-06-11:
There's also the `dpms-standby` module that may come into picture on your setup. And if so, you can try disabling that, like the example mentioned above. Give that you mention 5 minutes, that is very likely the one causing the symptom.

FrostKnight, 2021-06-12:
You might be right, hmmm.... how do I disable that module if thats the problem?

FrostKnight, 2021-06-12:
I just checked, it may not be your bug, because, it locks even when I am plugged in after 5 mins. I have no idea why though...

rickysarraf, 2021-06-12:
> I just checked, it may not be your bug, because, it locks even when I am plugged in after 5 mins. I have no idea why though...

Usually, on X, the server takes in some defaults. There's also the window manager that sets some of its own defaults. Maybe monitor the state of the system with `execsnoop` tool. That may give in some insight. 2 variants of the `execsnoop` tool are available in Debian: `execsnoop-bpfcc` and `execsnoop-perf`

FrostKnight, 2021-06-14:
Yeah.... I looked and, I found out I need to do this:

xset s off  -dpms

I think may be the answer I need... 

So yes, your right, this is no doubt a default issue. 

Once I do this, I assume that by pressing xset -s off -dpms, 

That it won't happen anymore in combination with xss-lock right?

Aka the issue I mentioned of blanking every 5 mins right?

rickysarraf, 2021-06-14:
I have no clue if that'd happen again or not. It all depends on what stack you are running. `xset` is basically you telling the X to do something. But there are others than you, who are on your machine. Your Window Manager may make calls at certain intervals. Or any other relevant tool for that matter.

Nevertheless, this isn't a `Laptop Mode Tools` issue so I'm closing this bug report. Thank you for investigating and confirming.

closed 2021-06-14 by rickysarraf