rickysarraf/laptop-mode-tools #174: laptop-mode-tools changes nvidia's prime D3 power management.

gulafaran, 2021-12-08:
i havent exactly figured out what bits of laptop-mode-tools thats changing it but upon reboot or plugging in or out the power cord changes it.

/sys/bus/pci/devices/0000:01:00.0/power/control gets changed to "on" instead of "auto" meaning its constantly powered on. so currently i have to "echo auto | sudo tee /sys/bus/pci/devices/0000:01:00.0/power/control" on boot/AC/Battery changes. doesnt occur without laptop-mode-tools.

rickysarraf, 2021-12-09:
That would be the `runtime-pm` module. Here's the specific [line](https://github.com/rickysarraf/laptop-mode-tools/blob/9839bd2a1d761f8fe476fa91aee2cc7909d4aa59/usr/share/laptop-mode-tools/modules/runtime-pm#L184)

`auto` is used when wanting to conserve power. OTOH, `on` is used when you want to run the device full power, which is the case when on AC.

So, as you mentioned, on power plug in, it'll be `on`.

gulafaran, 2021-12-09:
thanks, il modify that and add simply a line that sets nvidia to auto on AC aswell then. its mostly just generating heat and powers on by itself anyways when ran with prime-run so.

rickysarraf, 2021-12-10:
You could also try blacklisting that particular device id in
`runtime-pm.conf`.

s3nt fr0m a $martph0ne, excuse typ0s

gulafaran, 2021-12-10:
does that mean adding "0000:01:00.0" to the blacklist or do i use "01:00.0" from lspci?

rickysarraf, 2021-12-13:
On Fri, 2021-12-10 at 04:01 -0800, Tom Englund wrote:
> does that mean adding "0000:01:00.0" to the blacklist or do i use
> "01:00.0" from lspci?

I think the lspci device ids should work. Please try it out.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
Necessity is the mother of invention
आव्श्यक्ता आविष्कार की जननी है