rickysarraf/laptop-mode-tools #177: I can't do anything :(

LorenzoVasi, 2022-02-28:
**Describe the bug**
When i try to modify any option, it doens't work and it makes the following error:
Couldn't apply all request settings
I try to uncheck all settings because i thinked that it was my laptop, but it makes always this error

Applications logs with DEBUG, VERBOSE_OUTPUT enabled:
```bash
Exec with pkexec
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Failed in populateValues() when operating on /etc/laptop-mode/conf.d/board-specific
Couldn't apply setting for checkBox_board-specific
```

**To Reproduce**
//

**Expected behavior**
//

**Screenshots**
[image]

**Important Information:**
 - Distribution - Debian11 Bullseye
 - Linux Kernel - Linux 5.10.0-11-amd64
 - Laptop Mode Tools version - lastest
 - Desktop Environment, if any - GNOME41 with Wayland
 - Relevant System Logs - Up
 - Laptop Mode Tools verbose logs - Enabled

**Additional context**
//

Thx for all, have a good day :)

rickysarraf, 2022-03-02:
Thank you for reporting this issue. I can confirm the bug.

LorenzoVasi, 2022-03-02:
No problem, thank u for all your work 😊