rickysarraf/laptop-mode-tools #180: Touchpad and Touch Screen Issue in KDE

pasmon79, 2022-03-11:
Sometimes on restart my touchscreen and touch-pad fails to load. 

Can laptop-mode-tools be contributing to this. 

Kindly indicate the settings that affect these two items on my linux plasma.

rickysarraf, 2022-03-11:
Possibly. The `runtime-pm` module can hinder such devices. The best way to confirm that is by disabling `laptop-mode-tools` in `/etc/laptop-mode/laptop-mode.conf` by setting variable `ENABLE_LAPTOP_MODE_TOOLS=0` and the reboot.

Please do share the results and then we can look further into the cause.

pasmon79, 2022-03-11:
I already unchecked all the options in Laptop-mode, uninstalled it and removed /etc/laptop-mode completely. 
Yet some signatures remained, one of which disabled my wifi completely. I had to edit the default-wifi-powersave-on to take care of it.

I am absolutely sure that laptop-mode has edited some essential system file "not part of laptop-mode" conflicting with touch drivers.

rickysarraf, 2022-03-11:
> I already unchecked all the options in Laptop-mode, uninstalled it and removed /etc/laptop-mode completely. Yet some signatures remained, one of which disabled my wifi completely. I had to edit the [...]

Then you have a different problem. Possibly directly related to your device/driver. Nothing that `laptop-mode` does is persisted across reboots.

pasmon79, 2022-03-11:
Kindly guide me if laptop-mode tools has written over the files involving touch devices or not.

Thanks for your kind guidance.

rickysarraf, 2022-03-11:
> Kindly guide me if laptop-mode tools has written over the files involving touch devices or not.
> 

No. You must be having a different problem altogether. There's nothing here, from what you describe, that could now be caused by `laptop-mode-tools`. Once it is disabled, there's nothing it does.