rickysarraf/laptop-mode-tools #182: LCD brightness changes on USB insertion

LRitzdorf, 2022-04-04:
**Describe the bug**
When a USB device is plugged in, the LCD display brightness is reset to the level specified in `/etc/laptop-mode/conf.d/lcd-brightness.conf`.

This occurs with a Logitech Unifying dongle, flash drives, and unusual devices like a Digilent Analog Discovery (a USB-based oscilloscope multitool) and an Intel/Altera DE10-Lite FPGA development board. Thus, it seems to be triggered by all USB devices.

In #94, you note that "LMT is invoked whenever there is a power state change. It also is invoked during swsusp resume and *on USB device additions*" (emphasis mine). Is this expected behavior, then? If so, is there a way to prevent my brightness from being reset every time I connect a USB device? During everyday use, having my brightness reset whenever I connect a USB device is a bit inconvenient, as well as unintuitive.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit `lcd-brightness.conf` to specify your preferred backlight levels for each power state
1. Change your brightness manually to a different level
1. Plug in a USB device
1. Observe that brightness is reset to the level specified in the config file

**Expected behavior**
From an everyday use perspective, I would not expect connecting a USB device to trigger a brightness change, though it seems this might be intended behavior (see above).

**Important Information:**
 - Distribution: Arch Linux
 - Linux Kernel: `5.17.1-arch1-1`
 - Laptop Mode Tools version: `1.74-1` (AUR)
 - Desktop Environment: i3

rickysarraf, 2022-04-05:
Yes. This is the expected behavior. But you can customize it, if you don't like it.

```
$ cat /lib/udev/rules.d/99-laptop-mode.rules 
ACTION=="change", SUBSYSTEM=="power_supply", ENV{POWER_SUPPLY_NAME}=="|AC|ACAD|ADP*", RUN+="lmt-udev auto"
ACTION=="add|remove", SUBSYSTEM=="machinecheck", RUN+="lmt-udev auto"
ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force"

# Run a particular module only
#ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force modules=runtime-pm devices=%k"
```

You could change `lmt-udev force` to `auto`. But it may have some other side-effects like the newly plugged usb device may not inherit the power savings settings that you may have specified in laptop-mode-tools configs.

On the other hand, you could also enable the `module only` configuration:
```
ACTION=="add", SUBSYSTEM=="usb", RUN+="lmt-udev force modules=runtime-pm devices=%k"
```

Please do share your results. 🙏🏽

rickysarraf, 2022-04-05:
Yes. That was the reason to call `force`. see commit id: f4d6b93e33bb7a1fd030f073fb1bb48c0e60f4ba

LRitzdorf, 2022-04-06:
Perfect, that seems to work as expected! I suppose there's a reason that line is provided in the udev rules file :grin: (and I should probably have checked that before opening an issue here). Thanks for the quick response, and have a nice day!

closed 2022-04-06 by LRitzdorf