rickysarraf/laptop-mode-tools #198: Laptop Mode Tools turns on both front and back camera LEDs when battery is plugged in

JeanSchoeller, 2023-02-28:
When the battery is plugged in, the cameras LED of my laptop are turned on.

Specs:

Laptop: Surface Pro 5
Desktop environment: Hyprland
Linux kernel: linux-surface
Distribution: Arch

Steps to reproduce the behavior:
Install laptop-mode-tools from pacman
Enable laptop-mode-tools services
Both camera become enabled when the battery is plugged in. Disabling the services removes this behavior.

rickysarraf, 2023-03-17:
>When the battery is plugged in, the cameras LED of my laptop are turned on.

Battery plugged in ? Or the other way around, when AC plugged in ?

Is there nothing erratic reported in the kernel logs, worth mentioning here ?
I don't have the hardware so you'll have to help debug this issue.

We don't have any specific module for `camera` so my guess is that it is the `runtime-pm` subsystem triggering in power saving, and having issues with it. But this is just a guess.

How is the camera device enumerated under `sysfs` ?

JeanSchoeller, 2023-04-01:
This has been resolved. Disabling runtime-pm fixed it. Thank you. 
For info, the issue was when the AC was plugged in, there was no dmesg related to this when plugging it in.

closed 2023-04-01 by JeanSchoeller