rickysarraf/laptop-mode-tools #199: laptop-mode-tools acts as if the power status is inverted.

MikeChunko, 2023-03-16:
**Describe the bug**
As the title says: When I'm on AC power, my laptop acts as if it was on battery and everything slows way down. When I'm on battery, everything runs as it should. It's worth noting that "AC Power" mode runs slower than Battery Mode ever used to.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Laptop Mode Tools 1.74
2. Connect to AC power
3. Run `watch cpupower frequency-info` and observe the behavior
2. Disconnect from AC power
3. Run `watch cpupower frequency-info` and observe the behavior
5. Note that on AC power, the CPU while under a workload is much lower than on battery power. This behavior is also evident in applications running slower under AC power than on battery power

**Expected behavior**
CPU frequency should trend higher on AC power than on battery power

**Important Information:**
 - Distribution
    - Manjaro
 - Linux Kernel
    - 6.1.19-1
 - Laptop Mode Tools version
    - 1.74-1
 - Desktop Environment, if any
   - X11
 - Relevant System Logs
 - Laptop Mode Tools verbose logs
   - Please guide me to where I can find these, as there's nothing under `/var/log/`

**Additional context**
The problem first appeared when I upgraded from version 1.72.2 to 1.74-1. Downgrading back to 1.72.2 did not fix the issue.

I've confirmed via `acpi -a` as well as via `laptop_mode status` that my AC power status is being correctly detected

rickysarraf, 2023-03-17:
What is the below setting for you in  `/etc/laptop-mode/laptop-mode.conf` ?

```
#
# Enable laptop mode power savings, even when on AC power.
# This is useful when running as a headless machine, in low power mode
#
ENABLE_LAPTOP_MODE_ON_AC=0
```

MikeChunko, 2023-03-17:
> What is the below setting for you in `/etc/laptop-mode/laptop-mode.conf` ?
> 
> ```
> #
> # Enable laptop mode power savings, even when on AC power.
> # This is useful when running as a headless machine, in low power mode
> #
> ENABLE_LAPTOP_MODE_ON_AC=0
> ```

`ENABLE_LAPTOP_MODE_ON_AC=0`

I've disabled several of the optimizations in my quest to isolate this issue to laptop-mode, but here's my full config file:
```###############################################################################
#
# Configuration for Laptop Mode Tools
# -----------------------------------
#
# There is a "system" to the configuration setting names:
#    CONTROL_something=0/1   Determines whether Laptop Mode Tools controls 
#                            something
#    LM_something=value      Value of "something" when laptop mode is active
#    NOLM_something=value    Value of "something" when laptop mode is NOT
#                            active
#    AC_something=value      Value of "something" when the computer is running
#                            on AC power
#    BATT_something=value    Value of "something" when the computer is running
#                            on battery power
#
# There can be combinations of LM_/NOLM_ and AC_/BATT_ prefixes, but the
# available prefixes are different for each setting. The available ones are 
# documented in the manual page, laptop-mode.conf(8). If there is no LM_/
# NOLM_ in a setting name, then the value is used independently of laptop
# mode state, and similarly, if there is no AC_/BATT_, then the value is used
# independently of power state.
#
# Some options only work on ACPI systems. They are marked ACPI-ONLY.
#
# Note that this configuration file is a fragment of shell script: you
# can use all the features of the shell scripting language to achieve your
# desired configuration.
#
# 
# Modules
# -------
#
# Laptop Mode Tools modules have separate configuration files, that can be
# found in /etc/laptop-mode/conf.d. Please look through these configuration
# files as well, there are many useful power saving tools in there!
#
###############################################################################

###############################################################################
# Enable/Disable laptop-mode-tools execution
# ------------------------------------------
# Set it to 0 to completely disable laptop-mode-tools from running
###############################################################################
#
ENABLE_LAPTOP_MODE_TOOLS=1

###############################################################################
# Configuration debugging
# -----------------------
###############################################################################

#
# Set this to 1 if you want to see a lot of information when you start/stop 
# laptop_mode.
#
VERBOSE_OUTPUT=0
[... 270 lines cut ...]
# kernel default settings -- don't touch these unless you know what you're 
# doing.
#
DEF_UPDATE=5
DEF_XFS_AGE_BUFFER=15
DEF_XFS_SYNC_INTERVAL=30
DEF_XFS_BUFD_INTERVAL=1
DEF_MAX_AGE=30

#
# This must be adjusted manually to the value of HZ in the running kernel
# on 2.4, until the XFS people change their 2.4 external interfaces to work in
# centisecs. This can be automated, but it's a work in progress that still
# needs some fixes. On 2.6 kernels, XFS uses USER_HZ instead of HZ for
# external interfaces, and that is currently always set to 100. So you don't
# need to change this on 2.6.
#
XFS_HZ=100

#
# Seconds laptop mode has to to wait after the disk goes idle before doing
# a sync.
#
LM_SECONDS_BEFORE_SYNC=2
```