rickysarraf/laptop-mode-tools #202: lm-profiler not working on linux kernel 5.14 and higher

chadek, 2023-06-09:
**Describe the bug**
lm-profiler is not working on debian kernel 6.1.0-0.deb11.7-amd64 because of /proc/sys/vm/block_dump

**To Reproduce**
Steps to reproduce the behavior:
1. install laptop-mode-tools on a machine with a kernel 5.14 or higher 
2. run lm-profiler
3. The following error show up /usr/sbin/lm-profiler: 82: cannot create /proc/sys/vm/block_dump: Directory nonexistent

**Expected behavior**
lm-profiler should start profiling

**Important Information:**
 - Debian
 - kernel 6.1.0-0.deb11.7-amd64
 - Installed: 1.74-1.1

**Additional context**
As describe [here](https://patchwork.kernel.org/project/linux-block/cover/20210313030146.2882027-1-yi.zhang@huawei.com/#24166143), block_dump was removed from kernel, they suggest to use tracepoints instead. Maybe looking at /sys/kernel/tracing/README might help to find a replacement to the old block_dump interface

rickysarraf, 2023-08-16:
Thank you for the report. And my apologies for the delayed response 🙏🏽 

I will check on the status and possible alternates.

rickysarraf, 2023-08-25:
@chadek Does [this](https://github.com/rickysarraf/laptop-mode-tools/pull/205) look good to you ?

chadek, 2023-08-26:
Just tried it out and it seams to work. I had no clue on how this tool work normally as it is the first time I use it. Profiling starts and seams to do the job. It would just be better if this tool as less dependency (btrace  come with a lot a things) but this is an other topics.

Thanks for the fix!

rickysarraf, 2023-08-28:
> Just tried it out and it seams to work. I had no clue on how this tool work normally as it is the first time I use it. Profiling starts and seams to do the job. It would just be better if this tool [...]

Thanks for the feedback. The name being duplicate is a possibility, which is why that is pushed to the packagers, who could add a dependency/recommendation to this package.

closed 2023-08-28 by rickysarraf