rickysarraf/laptop-mode-tools #203: can not start a new program in exec-commands module

jyf-111, 2023-08-16:
**Describe the bug**

i hope to start `polybar` and `picom` in `NOLM_AC_EXEC_COMMAND_0`,
but no program start when battery mode change,
and i am confirm that the `EXEC_COMMAND` is execuated

**To Reproduce**

```conf
# exec-commands.conf
BATT_EXEC_COMMAND_0="echo batt >> /home/jyf/log.log && pkill -9 polybar"
LM_AC_EXEC_COMMAND_0=""
NOLM_AC_EXEC_COMMAND_0="echo ac >> /home/jyf/log.log && polybar"
```

exec `tail -f /home/jyf/log.log` to confirm the command is execuated

```text
batt
ac
batt
ac
batt
ac
batt
ac
```
but polybar cant start

**Important Information:**
 - Distribution:   manjaro
 - Linux Kernel:   6.1.44-1-MANJARO 
 - Laptop Mode Tools version:  1.74
 - Desktop Environment, if any:  i3wm

rickysarraf, 2023-08-16:
What is polybar ?

s3nt fr0m a $martph0ne, excuse typ0s

jyf-111, 2023-08-16:
[polybar](https://github.com/polybar/polybar）is a status bar for i3wm.

and i have double checked my command is correct.

rickysarraf, 2023-08-16:
laptop-mode-tools is run as root whereas you'd want polybar run as your
current desktop user.

On X11, this could be worked around but not a recommended thing to do. On
Wayland, I haven't tried.

s3nt fr0m a $martph0ne, excuse typ0s

jyf-111, 2023-08-16:
i see. so we can not start a GUI program in `XX_EXEC_COMMAND_X` as a normal user. 
am i right?

rickysarraf, 2023-08-16:
> i see. so we can not start a GUI program in `XX_EXEC_COMMAND_X` as a normal user. am i right?

Yes. That cannot be done. Given this is not an issue and working as designed, I'm closing this issue. Please feel free to re-open if you think otherwise.

closed 2023-08-16 by rickysarraf