rickysarraf/laptop-mode-tools #204: bug in acpi/udev integration

Shea690901, 2023-08-16:
**Describe the bug**
For acpi events (ac_adaptor|battery|button[ /]lid).* three different scripts are called.
Each of these scripts calls ```/lib/udev/lmt-udev``` without any argument, effectively doing the same.

The called udev script, when not getting "auto" as it's first argument always, regardless of running on a system using systemd or not, uses the internal function ```non_systemd_way```, here again without any argument, and this function itself then calls ```/usr/sbin/laptop_mode```, still without any arguments....

When called via udev rules it isn't much better:
Here we have 2 rules which on a systemd-system reload-or-restart ```laptop-mode.service``` and 1 rule regardless of the used init-system calling ```/usr/sbin/laptop_mode "force"```.
If we have no ```/lib/udev/hotplug.functions``` we might get an error message that ```force did not appear before the timeout!``` if ```/usr/sbin/laptop_mode``` can't be found within 120s, when called via acpid the same error message would be missing the word "force".

**To Reproduce**
Steps to reproduce the behavior:
Install laptop-mode-tools.
Checked for version 1.74.

**Expected behavior**
A correct and logical integration with both acpid and udevd and not just working (is it even completely working?) by mistake....

**Screenshots**
Just take a close look at the supplied scripts....

**Important Information:**
 - Distribution
    each using vanilla sources
 - Laptop Mode Tools version 1.74, perhaps earlier too, not checked...

rickysarraf, 2023-08-18:
What erratic behavior did you see in action ? Maybe some real logs will help ascertain your cause.

It has been working fine for me and many more users. This is the first report contradicting the impression otherwise.

Shea690901, 2023-09-24:
Sorry for the long delay, problems in setting up my new machine....

As I wrote:
It might work accidently, but if all is working as expected, according the code, it IS axcidently...
Please just take a CLOSE look at the code as I described in my report and not just read what you WANT done, but what is actual done, just as I described.
Everything is doing the same...
That might be what you want to do, but in that case the code volume could be halved without any problems....