rickysarraf/laptop-mode-tools #209: Error when read /sys/class/power_supply/BAT0/energy_full_design

RiKkzu, 2024-03-28:
>systemctl status laptop-mode.service
>× laptop-mode.service - Laptop Mode Tools
>    Loaded: loaded (/lib/systemd/system/laptop-mode.service; disabled; preset: enabled)
>     Active: failed (Result: exit-code) since Fri 2024-03-29 00:59:40 +05; 1h 22min ago
       Docs: man:laptop_mode(8)
             man:laptop-mode.conf(8)
             http://github.com/rickysarraf/laptop-mode-tools
>    Process: 841 ExecStartPre=/bin/rm -f /var/run/laptop-mode-tools/enabled (code=exited, status=0/SUCCESS)
    Process: 857 ExecStartPre=/bin/rm -f /var/run/laptop-mode-tools/state (code=exited, status=0/SUCCESS)
    Process: 864 ExecStart=/usr/sbin/laptop_mode init force (code=exited, status=2)
    Process: 981 ExecStopPost=/bin/rm -f /var/run/laptop-mode-tools/enabled (code=exited, status=0/SUCCESS)
    Process: 987 ExecStopPost=/bin/rm -f /var/run/laptop-mode-tools/state (code=exited, status=0/SUCCESS)
   Main PID: 864 (code=exited, status=2)
        CPU: 95ms
>29 00:59:39 debian2 systemd[1]: Starting laptop-mode.service - Laptop Mode Tools...
29 00:59:40 debian2 laptop_mode[978]: cat: /sys/class/power_supply/BAT0/energy_full_design: No such device
29 00:59:40 debian2 systemd[1]: laptop-mode.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
29 00:59:40 debian2 laptop_mode[864]: /usr/sbin/laptop_mode: 793: [: Illegal number:
29 00:59:40 debian2 laptop_mode[864]: /usr/sbin/laptop_mode: 812: [: Illegal number:
29 00:59:40 debian2 laptop_mode[864]: /usr/sbin/laptop_mode: 816: arithmetic expression: expecting primary: "42000000 * 100 / "
29 00:59:40 debian2 systemd[1]: laptop-mode.service: Failed with result 'exit-code'.
29 00:59:40 debian2 systemd[1]: Failed to start laptop-mode.service - Laptop Mode Tools.

>ls -l /sys/class/power_supply/BAT0/
-rw-r--r-- 1 root root 4096 мар 29 01:07 alarm
-r--r--r-- 1 root root 4096 мар 29 00:59 capacity
-r--r--r-- 1 root root 4096 мар 29 01:07 capacity_level
-r--r--r-- 1 root root 4096 мар 29 00:59 cycle_count
lrwxrwxrwx 1 root root    0 мар 29 01:07 device -> ../../../PNP0C0A:00
-r--r--r-- 1 root root 4096 мар 29 00:59 energy_full
-r--r--r-- 1 root root 4096 мар 29 00:59 energy_full_design
-r--r--r-- 1 root root 4096 мар 29 00:59 energy_now
drwxr-xr-x 3 root root    0 мар 29 00:59 hwmon1
-r--r--r-- 1 root root 4096 мар 29 00:59 manufacturer
-r--r--r-- 1 root root 4096 мар 29 00:59 model_name
drwxr-xr-x 2 root root    0 мар 29 01:07 power
-r--r--r-- 1 root root 4096 мар 29 00:59 power_now
-r--r--r-- 1 root root 4096 мар 29 00:59 present
-r--r--r-- 1 root root 4096 мар 29 00:59 serial_number
-r--r--r-- 1 root root 4096 мар 29 00:59 status
lrwxrwxrwx 1 root root    0 мар 29 00:59 subsystem -> ../../../../../../../../../class/power_supply
-r--r--r-- 1 root root 4096 мар 29 00:59 technology
-r--r--r-- 1 root root 4096 мар 29 00:59 type
-rw-r--r-- 1 root root 4096 мар 29 00:59 uevent
-r--r--r-- 1 root root 4096 мар 29 00:59 voltage_min_design
-r--r--r-- 1 root root 4096 мар 29 00:59 voltage_now

>cat /sys/class/power_supply/BAT0/uevent
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Unknown
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_VOLTAGE_NOW=7400000
POWER_SUPPLY_POWER_NOW=4158000
POWER_SUPPLY_ENERGY_FULL=50000000
POWER_SUPPLY_ENERGY_NOW=30000000
POWER_SUPPLY_CAPACITY=60
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=SR Real Battery
POWER_SUPPLY_MANUFACTURER=Intel SR 1
POWER_SUPPLY_SERIAL_NUMBER=123456789

>uname -a
Linux debian2 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux

Notebook Prestigio 133S

rickysarraf, 2025-02-10:
That hardware doesn't report via the standard interface.

```
/sys/class/power_supply/BAT0/energy_full_design
```

Do you have insight into what alternative interface it has, if any ?

rickysarraf, 2025-02-10:
> That hardware doesn't report via the standard interface.
> 
> ```
> /sys/class/power_supply/BAT0/energy_full_design
> ```
> 
> Do you have insight into what alternative interface it has, if any ?

And the second output about BAT0 report it fine.

```
ls -l /sys/class/power_supply/BAT0/
-rw-r--r-- 1 root root 4096 мар 29 01:07 alarm
-r--r--r-- 1 root root 4096 мар 29 00:59 capacity
-r--r--r-- 1 root root 4096 мар 29 01:07 capacity_level
-r--r--r-- 1 root root 4096 мар 29 00:59 cycle_count
lrwxrwxrwx 1 root root 0 мар 29 01:07 device -> ../../../PNP0C0A:00
-r--r--r-- 1 root root 4096 мар 29 00:59 energy_full
-r--r--r-- 1 root root 4096 мар 29 00:59 energy_full_design
-r--r--r-- 1 root root 4096 мар 29 00:59 energy_now
drwxr-xr-x 3 root root 0 мар 29 00:59 hwmon1
-r--r--r-- 1 root root 4096 мар 29 00:59 manufacturer
-r--r--r-- 1 root root 4096 мар 29 00:59 model_name
drwxr-xr-x 2 root root 0 мар 29 01:07 power
-r--r--r-- 1 root root 4096 мар 29 00:59 power_now
-r--r--r-- 1 root root 4096 мар 29 00:59 present
-r--r--r-- 1 root root 4096 мар 29 00:59 serial_number
-r--r--r-- 1 root root 4096 мар 29 00:59 status
lrwxrwxrwx 1 root root 0 мар 29 00:59 subsystem -> ../../../../../../../../../class/power_supply
-r--r--r-- 1 root root 4096 мар 29 00:59 technology
-r--r--r-- 1 root root 4096 мар 29 00:59 type
-rw-r--r-- 1 root root 4096 мар 29 00:59 uevent
-r--r--r-- 1 root root 4096 мар 29 00:59 voltage_min_design
-r--r--r-- 1 root root 4096 мар 29 00:59 voltage_now
```

So something is wrong in your setup.