rickysarraf/laptop-mode-tools #211: laptop-mode-tools Not finding commands to run

rohitnanda1443, 2025-02-08:
I am trying laptop-mode-tools application and when I start it the below are the errors. The PC is an AMD Athlon II.

`# rc-service laptop_mode start
 * Starting laptop_mode ...
/etc/laptop-mode/laptop-mode.conf: line 111: ac97-powersave: command not found
/etc/laptop-mode/laptop-mode.conf: line 112: cpufreq: command not found
/etc/laptop-mode/laptop-mode.conf: line 113: dpms-standby: command not found
/etc/laptop-mode/laptop-mode.conf: line 116: exec-commands: command not found
/etc/laptop-mode/laptop-mode.conf: line 117: hal-polling: command not found
/etc/laptop-mode/laptop-mode.conf: line 118: hdparm: command not found
/etc/laptop-mode/laptop-mode.conf: line 121: nmi-watchdog: command not found
/etc/laptop-mode/laptop-mode.conf: line 122: pcie-aspm: command not found
/etc/laptop-mode/laptop-mode.conf: line 123: runtime-pm: command not found
/etc/laptop-mode/laptop-mode.conf: line 124: sched-mc-power-savings: command not found
/etc/laptop-mode/laptop-mode.conf: line 125: sched-smt-power-savings: command not found
/etc/laptop-mode/laptop-mode.conf: line 126: terminal-blanking: command not found
/etc/laptop-mode/laptop-mode.conf: line 127: wireless-ipw-power: command not found
/etc/laptop-mode/laptop-mode.conf: line 128: wireless-iwl-power: command not found
/etc/laptop-mode/laptop-mode.conf: line 129: wireless-power: command not found
/etc/laptop-mode/laptop-mode.conf: line 111: ac97-powersave: command not found
/etc/laptop-mode/laptop-mode.conf: line 112: cpufreq: command not found
/etc/laptop-mode/laptop-mode.conf: line 113: dpms-standby: command not found
/etc/laptop-mode/laptop-mode.conf: line 116: exec-commands: command not found
/etc/laptop-mode/laptop-mode.conf: line 117: hal-polling: command not found
/etc/laptop-mode/laptop-mode.conf: line 118: hdparm: command not found
/etc/laptop-mode/laptop-mode.conf: line 121: nmi-watchdog: command not found
/etc/laptop-mode/laptop-mode.conf: line 122: pcie-aspm: command not found
/etc/laptop-mode/laptop-mode.conf: line 123: runtime-pm: command not found
/etc/laptop-mode/laptop-mode.conf: line 124: sched-mc-power-savings: command not found
/etc/laptop-mode/laptop-mode.conf: line 125: sched-smt-power-savings: command not found
/etc/laptop-mode/laptop-mode.conf: line 126: terminal-blanking: command not found
/etc/laptop-mode/laptop-mode.conf: line 127: wireless-ipw-power: command not found
/etc/laptop-mode/laptop-mode.conf: line 128: wireless-iwl-power: command not found
/etc/laptop-mode/laptop-mode.conf: line 129: wireless-power: command not found
ERROR: hdparm not installed.
ERROR: hdparm not installed.                                              [ ok ]`

I have edited these 3 files below which shows md5 checksum error. How does one resolve this and / or edit the config files:

`!!! /etc/laptop-mode/laptop-mode.conf has incorrect MD5sum
!!! /etc/laptop-mode/conf.d/nouveau.conf has incorrect MD5sum
!!! /etc/laptop-mode/conf.d/cpufreq.conf has incorrect MD5sum
* Checking app-laptop/laptop-mode-tools-1.74 ...`

rickysarraf, 2025-02-09:
What distribution are you on ? What is the version of laptop mode tools you
are using?

s3nt fr0m a $martph0ne, excuse typ0s

rohitnanda1443, 2025-02-09:
I am on Gentoo linux and using version 1.74.

has the whole diff of the files: https://forums.gentoo.org/viewtopic-p-8854176.html#8854176

rickysarraf, 2025-02-10:
```
/etc/laptop-mode # diff -u laptop-mode.conf laptop-mode.conf.old
--- laptop-mode.conf    2025-02-08 08:11:57.452538454 +0530
+++ laptop-mode.conf.old        2025-01-26 22:27:36.885402428 +0530
@@ -59,7 +59,7 @@
 VERBOSE_OUTPUT=0
 
... snipped ... 
 
 #
@@ -107,25 +108,25 @@
 #
 # List of modules which can be automatically enabled with this setting are:
 #
-# ac97-powersave
-# cpufreq
-# dpms-standby
+ ac97-powersave
+ cpufreq
+ dpms-standby
 # eee-superhe
-# ethernet
-# exec-commands
```

Your entire installation/configuration for laptop-mode-tools is broken. I don't know how you got into this soup.

The configuration files in /etc/laptop-mode/laptop-mode.conf are expected to be sources by the primary `laptop_mode` script. They are sourced only for variable reading. By uncommenting the modules, you make then behave like a shell script, which is why you get those errors like:

```
/etc/laptop-mode/laptop-mode.conf: line 111: ac97-powersave: command not found
```

You should keep everything in configuration files commented. And only fiddle with the variables' assignment there.

Also, check out the fine `laptop-mode.conf (8)` manual page.

rohitnanda1443, 2025-02-23:
Thanks Ricky, If I use the original config files the laptop-mode-tools start but my CPU is not scaled down. How do I enforce those configs ?

rickysarraf, 2025-02-23:
Please provide some details. Maybe run laptop-mode-tools in verbose mode

s3nt fr0m a $martph0ne, excuse typ0s

rickysarraf, 2025-02-23:
Also check if the cpufreq module is enabled

s3nt fr0m a $martph0ne, excuse typ0s