rickysarraf/laptop-mode-tools #212: Add option to manage `/sys/firmware/acpi/platform_profile`

ikus060, 2025-02-09:
In recent Linux Kernel their is a new ACPI-based power profile interface introduced in newer kernel versions (Linux 5.12+). It allows selecting power modes like performance, balanced, or low-power, depending on the laptop's firmware support.

On my system when Ichange it from "balance" to "quiet", I get reduce the power usage from ~0.8w.

Could we consider adding this option to LMT ?

rickysarraf, 2025-02-10:
> > In recent Linux Kernel their is a new ACPI-based power profile interface introduced in newer kernel versions (Linux 5.12+). It allows selecting power modes like performance, balanced, or low-power, depending on the laptop's firmware support.
> 
> I guess you meanth 6.12+ ?
> 
> > On my system when Ichange it from "balance" to "quiet", I get reduce the power usage from ~0.8w.
> 

On my machine, it reports:

```
⛢ 20:22:05 rrs@priyasi ~  
@ cat /sys/firmware/acpi/platform_profile
balanced

ॐ 20:22:08 rrs@priyasi ~  
@ cat /sys/firmware/acpi/platform_profile_choices
low-power balanced performance

♅ 20:22:10 rrs@priyasi ~  
@ 
```

I'm missing `quiet`. But more than that, it'd be nice to know what all possible choices are made available.

ikus060, 2025-02-10:
@rickysarraf According to my research, platform_profile_choices returns a list of available profiles. The values depend on how ACPI reports the available profiles. The values are different but similar across vendors. According to the kernel source code, it seems that the list returned by /sys/firmware/acpi/platform_profile_choices is ordered from lower power consumption to higher performance.

Here the list of possible values according to kernel source code:
https://github.com/torvalds/linux/blob/a64dcfb451e254085a7daee5fe51bf22959d52d3/drivers/acpi/platform_profile.c#L27