rickysarraf/laptop-mode-tools #215: Should we move the Debian packaging to Salsa?

tillea, 2025-07-19:
Hi,
I realized that you are maintaining the Debian packaging here at Github.  As you might have seen the Debian package itself moved on with meanwhile four NMUs.  I would suggest to move the Debian packaging to salsa.debian.org/debian to enable Debian maintainers doing team uploads of the package smoothly.  I would volunteer to migrate the `master` tree of this repository to the Debian team and add the changes of the NMU.  I would also volunteer to do some upload incorporating the new Vcs fields.
What do you think?
Kind regards, Andreas.

rickysarraf, 2025-07-21:
Hello Andreas,

Sounds good to me. Please just add me as a maintainer as well.

Thanks,
Ritesh

rickysarraf, 2025-07-25:
@tillea Thank you so much for caring for this tool. All looks good to me.

> I'm wondering whether we should upload the package as is once trixie is released or whether you might like to inject the latest upstream version.

Lets upload to Debian, post Trixie, as it is with the work you've put to revive it.

Later, as time permits, I'll follow-up with a new upstream

tillea, 2025-07-25:
OK, I'll do post-Trixie release.  Thanks a lot for your quick response in any case.

tillea, 2025-07-27:
Am Fri, Jul 25, 2025 at 05:59:29AM -0700 schrieb Ritesh Raj Sarraf:
> Lets upload to Debian, post Trixie, as it is with the work you've put to revive it.

OK, uploaded to delayed=15.
 
> Later, as time permits, I'll follow-up with a new upstream

Cool! Thanks a lot, Andreas.